nlef/moonraker-telegram-bot #26: Consistent config format between

aka13-404, 2021-06-01:
It might make sense to unify the configuration format with klipper and moonraker. 

The current config format is not unified with klipper, which might lead to user input errors and undesired behaviour.

timelapse {
    heigth = 0
    enabled = true
}

The config in example for the timelapse module uses curved brackets, tabulation and the ability to disable a featur by setting a 
"enabled = true" flag.

With klipper and moonraker, a feature is enabled by having it present in the config in square brackets, i.e.:

[display_status]
[pause_resume]

The disabling of a feature is simply omitting it from the config alltogether.
Parameters are set to 0 per default, and unless entered into the config are inactive, as in 

[stepper_x]
gear_ratio: 5:1

Unless gear ratio is present in the config, it is set to 0 and not used.

Referring to the initial example with the timelapse module of the bot,
it should look like this:
[timelapse]
#height: 0.2

It should be considered, that should this change be implemented, it will break backwards compatibility with existing installations. Nevertheless, in my opinion, it is urgently needed to make this change while the userbase is small, and the available features are minimal. 

Input from @zztopper as well as @suglosta and @RushOnline is more than welcome.

zztopper, 2021-06-01:
Seems legit, I completely agree with @aka13-404.
We already had a conversation with @nlef on this, but he said he's used to this format and finds it convenient as he is a Java developer.
Also, this format makes any custom config additions and\or changes very hard to support in code (I tried once to add a section for gcode macros running).

nlef, 2021-06-01:
Ok. I'll rework config to be more like python configs & klipper/moonraker configuration logic.

aka13-404, 2021-06-13:
The following is a preview of the suggested config change:

```
[bot]
server: "localhost"
chat_id: "68266244"
bot_token: "1826261904:AAF1207YSq14OzlcECqultI59uvIH2rhVHc"
light_device: "chamber_led"
debug: true

[camera]
host: "http://192.168.1.3:8080/stream.mp4"
flipVertically: false
flipHorisontally: false
videoDuration: 125
light_control_timeout: 2

[progress_notification]
percent: 5
heigth: 5
time: 5
min_delay_between_notifications: 60

[timelapse]
basedir: "/mnt/stationary/timelapseusb"
cleanup: true
heigth: 0.2
time: 5
target_fps: 30	

[telegram_ui]
hidden_methods: "/gif", 
enabled_macros: "MANUAL_BED_LEVEL"
```

nlef, 2021-06-14:
@aka13-404 @zztopper maybe this should be closed? new config format was introduced in release 1.0.0

closed 2021-06-14 by aka13-404