nlef/moonraker-telegram-bot #28: Alternative triggers for timelapse start and end

aka13-404, 2021-06-04:
Since the priting is often done, before service operations take place (i.e. moving the table on bedslinger printers forward), it might be a good idea to be able to define alternative endpoints for the timelapse feature of the bot.

Also, since before the print some service operations, like a prime line, may take place, a alternative starting point for the timelapse might be interesting.

In addition, a timelapse pause function, for filament or toolchanges might be of value.

I suggest using klipper macros, specifically the RESPOND MSG G-Code.
`RESPOND PREFIX=timelapse MSG=start`. `RESPOND PREFIX=timelapse MSG=stop`, `RESPOND PREFIX=timelapse MSG=pause`, `RESPOND PREFIX=timelapse MSG=continue` respectively.

nlef, 2021-07-11:
implemented this in 78648c031d8c73408291d7402eafa7a83491e7a7
@aka13-404 pls check it)

aka13-404, 2021-07-30:
Working as intended since 1.1.0

closed 2021-07-30 by aka13-404