nlef/moonraker-telegram-bot #35: Feature: Notification on bed temperature reached after print completion

fishboy7296, 2021-06-16:
Would be nice to be notified when the print bed has cooled down to a specific temperature (specified in the configuration) after a print has completed.

nlef, 2021-07-11:
@fishboy7296 Try to use gcode bellow to sent custom notifications. Docs will be updated later
RESPOND PREFIX=tgnotify MSG="my super message"
RESPOND PREFIX=tgalarm MSG="my alarm"

nlef, 2021-07-18:
@fishboy7296 is this working for you?

fishboy7296, 2021-07-18:
Thanks for implementing this feature.  Unfortunately I won’t have access to
my printer for a few more weeks, but will test as soon as I can.

aka13-404, 2021-07-30:
We have tested this quite extensively, should be working now. Do reopen this if it does not.

closed 2021-07-30 by aka13-404

MCristoni, 2021-12-02:
Hope you're all doing well
The respond command isn't working for me
I've setup the following macro in my fluidd macros.cfg

`[gcode_macro PRE_HEAT_PLA]
#   Pre Heat PLA
gcode:
    M140 S60
    RESPOND PREFIX=tgalarm MSG="Setting bed temperature to: $bed_target"
    RESPOND PREFIX=tgalarm MSG="Current bed temperature: bed_temp"
    M104 S210
    RESPOND PREFIX=tgalarm MSG="Setting extruder temperature to: $extruder_target"
    RESPOND PREFIX=tgalarm MSG="Current extruder temperature: extruder_temp"`
    
   But I don't receive these messages on Telegram, I get a ` // Unknown command:"RESPOND"`
   in my Fluidd console instead

aka13-404, 2021-12-02:
Please open a new issue next time. 
As stated in the documentation, you obviously need the respond section (https://github.com/Klipper3d/klipper/blob/master/docs/Config_Reference.md#respond) enabled to use respond-type commands to write to console in klipper.