nlef/moonraker-telegram-bot #36: File is too big

breareros, 2021-06-18:
При загрузке больших файлов gcode (например 60Мб) загрузка файла не происходит.
В логе 
`2021-06-18 15:42:15,093 - __main__ - ERROR - Exception while handling an update:
Traceback (most recent call last):
  File "/home/pi/moonraker-telegram-bot-env/lib/python3.7/site-packages/telegram/ext/utils/promise.py", line 96, in run
    self._result = self.pooled_function(*self.args, **self.kwargs)
  File "/home/pi/moonraker-telegram-bot/main.py", line 364, in upload_file
    bio.write(doc.get_file().download_as_bytearray())
  File "/home/pi/moonraker-telegram-bot-env/lib/python3.7/site-packages/telegram/files/document.py", line 125, in get_file
    return self.bot.get_file(file_id=self.file_id, timeout=timeout, api_kwargs=api_kwargs)
  File "/home/pi/moonraker-telegram-bot-env/lib/python3.7/site-packages/telegram/bot.py", line 129, in decorator
    result = func(*args, **kwargs)
  File "/home/pi/moonraker-telegram-bot-env/lib/python3.7/site-packages/telegram/bot.py", line 2293, in get_file
    result = self._post('getFile', data, timeout=timeout, api_kwargs=api_kwargs)
  File "/home/pi/moonraker-telegram-bot-env/lib/python3.7/site-packages/telegram/bot.py", line 295, in _post
    f'{self.base_url}/{endpoint}', data=data, timeout=effective_timeout
  File "/home/pi/moonraker-telegram-bot-env/lib/python3.7/site-packages/telegram/utils/request.py", line 361, in post
    **urlopen_kwargs,
  File "/home/pi/moonraker-telegram-bot-env/lib/python3.7/site-packages/telegram/utils/request.py", line 279, in _request_wrapper
    raise BadRequest(message)
telegram.error.BadRequest: File is too big
`

Сам бот об ошибке не сообщает.

nlef, 2021-06-18:
Для ботов есть ограничение в 20мб для скачиваемых файлов. Сделаю поддержку распаковки сжатых файлов и логирование ошибок с уведомелнием пользователя.

nlef, 2021-06-28:
@breareros проверь. вроде работает)

breareros, 2021-06-29:
Отлично работает!!!

closed 2021-06-29 by breareros