nlef/moonraker-telegram-bot #37: monnraker authorization

breareros, 2021-06-23:
При добавлении пользователей через fluidd и установке в mooraker.conf:

```
[authorization]
force_logins: true
```

бот не будет работать.
Прошу добавить авторизацию бота в moonraker по логину/паролю.

DishersCave, 2021-09-26:
Не работает [timelapse]
Пробовал абсолютно разные настройки ...
текущий конфиг 

[timelapse]
basedir: /tmp/timelapse
copy_finished_timelapse_dir: /home/pi/timelapse/finished
cleanup: false
time: 5
target_fps: 15
manual_mode: false

p.s. файлы в папках не создает((

nlef, 2021-09-26:
> 
> 
> Не работает [timelapse]
> Пробовал абсолютно разные настройки ...
> текущий конфиг
> 
> [timelapse]
> basedir: /tmp/timelapse
> copy_finished_timelapse_dir: /home/pi/timelapse/finished
> cleanup: false
> time: 5
> target_fps: 15
> manual_mode: false
> 
> p.s. файлы в папках не создает((

Как проблема с таймлапсом связана с авторизацией? Я создал отдельный issue для обсуждения)

DishersCave, 2021-09-26:
Прошу прощения, не понял, что пишу как коммент, а не как новую тему )

closed 2021-11-06 by nlef