nlef/moonraker-telegram-bot #45: Не работает [timelapse]

nlef, 2021-09-26:
Не работает [timelapse]
Пробовал абсолютно разные настройки ...
текущий конфиг 

[timelapse]
basedir: /tmp/timelapse
copy_finished_timelapse_dir: /home/pi/timelapse/finished
cleanup: false
time: 5
target_fps: 15
manual_mode: false

p.s. файлы в папках не создает((

_Originally posted by @DishersCave in https://github.com/nlef/moonraker-telegram-bot/issues/37#issuecomment-927372167_

nlef, 2021-09-26:
@DishersCave
Просьба уточнить, а фото и видео бот успешно делает? Так же имеются ли у пользователя, от которого работает бот, права на запись в указанные в настройках каталоги?

DishersCave, 2021-09-26:
Да, фото и видео бот присылает: и по запросу, и с уведомлениями.
Насчёт последнего вопроса: я не знаю, как это проверить.

nlef, 2021-09-26:
@DishersCave 
При запуске печати в папке /tmp/timelapse не создается вообще ничего?
Просьба предоставить лог файл. telegram.log
По умолчанию лежит или в /tmp или в /home/pi/klipper_logs

DishersCave, 2021-09-26:
Нет. Даже папка сама не создалась, создавал её вручную.
Куда лог скинуть ?

nlef, 2021-09-26:
> Куда лог скинуть ?

https://pastebin.com/

DishersCave, 2021-09-26:
https://drive.google.com/file/d/1llx7IlKtJ24ymIzSwbM0U0ZVEyhyfF65/view?usp=sharing

Ссылка на лог

nlef, 2021-09-26:
@DishersCave 
В логе множество ошибок с подключением к сети. И включена авторизация в мунракере.
А какой слайсер используется? похоже отсутствуют данные по филаменту. Исправлю в ближайшее время.

DishersCave, 2021-09-27:
Это видимо старые ошибки, сейчас всё работает и авторизация отключена.
Слайсер cura

closed 2021-09-27 by nlef