nlef/moonraker-telegram-bot #46: Install script will break cinnamon desktop environments

th33xitus, 2021-09-27:
This purge command just broke my cinnamon desktop environment on my development VM:
https://github.com/nlef/moonraker-telegram-bot/blob/c2c115b6c04a428806d2d034da4c1d0a03ad7307/scripts/install.sh#L60

That command will remove:
`blueman* cinnamon* python3-pil*`

[image]

I won't add the bot installer to KIAUH until that is fixed. I have to assume that this can also break other users systems. Im not going to risk that by distributing that script via KIAUH.

aka13-404, 2021-09-27:
Alles klar, wird gemacht :D

nlef, 2021-09-27:
@th33xitus 
here's fix 14063c3b75a79e1d3d7cb12ee201939666a687db

th33xitus, 2021-09-28:
Fixed! 😄
Now my desktop environment stays alive hehe.

PR is also already merged 😄

closed 2021-09-28 by th33xitus