nlef/moonraker-telegram-bot #59: Is it possible to get a notification when the print starts?

simpat1zq, 2021-11-16:
Hi, I don't see it in the docs, so I'm not sure if it's possible, but how can I get a message when a print is started?

nlef, 2021-11-16:
@simpat1zq 
here is an example of a message that should come when printing starts 
[image]

there must be some errors in the log. please give me log

simpat1zq, 2021-11-16:
I just started a print and here's the output since then:  

`2021-11-16 16:37:31,464 - __main__ - ERROR - Exception while handling an update:
Traceback (most recent call last):
  File "/home/pi/moonraker-telegram-bot-env/lib/python3.7/site-packages/apscheduler/executors/base.py", line 125, in run_job
    retval = job.func(*job.args, **job.kwargs)
  File "/home/pi/moonraker-telegram-bot/bot/main.py", line 111, in send_print_start_info
    send_file_info(context.bot, notifier.silent_status, message)
  File "/home/pi/moonraker-telegram-bot/bot/main.py", line 115, in send_file_info
    message, bio = klippy.get_file_info(message)
  File "/home/pi/moonraker-telegram-bot/bot/klippy.py", line 244, in get_file_info
    img = Image.open(urlopen(f"http://{self._host}/server/files/gcodes/{urllib.parse.quote(self._thumbnail_path)}")).convert('RGB')
  File "/usr/lib/python3.7/urllib/request.py", line 222, in urlopen
    return opener.open(url, data, timeout)
  File "/usr/lib/python3.7/urllib/request.py", line 531, in open
    response = meth(req, response)
  File "/usr/lib/python3.7/urllib/request.py", line 641, in http_response
    'http', request, response, code, msg, hdrs)
  File "/usr/lib/python3.7/urllib/request.py", line 569, in error
    return self._call_chain(*args)
  File "/usr/lib/python3.7/urllib/request.py", line 503, in _call_chain
    result = func(*args)
  File "/usr/lib/python3.7/urllib/request.py", line 649, in http_error_default
    raise HTTPError(req.full_url, code, msg, hdrs, fp)
urllib.error.HTTPError: HTTP Error 404: Not Found
`

simpat1zq, 2021-11-17:
I did configure the notification settings to not send any notifications based on height, percent, or time. Could that have any effect on this? Also, when it sent me the completion notification, there was no picture attached. The pictures are working because it sent me pics when I had it configured with the defaults which sent a picture every 60 seconds. Thanks for your help with this.

nlef, 2021-11-17:
@simpat1zq 
This problem will be fixed in the next release. please wait a bit)

nlef, 2021-11-26:
@simpat1zq 
pls check new release

closed 2021-12-02 by aka13-404