nlef/moonraker-telegram-bot #65: Add the possibility to run shell script with/instead of sending timelapse video

zztopper, 2021-11-27:
If the timelapse video is greater than 50mb, it's not possible for bot to send it to telegram chat.
So that's why I want to send it to an external storage (cloud/local webserver, youtube etc) via shell script.
Shell script would take the timelapse file name with full path as an input parameter, and return the URL of the uploaded video, so that the bot could send the URL to the chat.

Sending timelapse video to the telegram chat as it is now should be optional, controlled with a param.
If the script name variable is not empty, script should be run after timelapse video is composed.

nlef, 2021-11-27:
@zztopper 
pls check implementation dc9fbb0027400f3d762efabf9f65f765d3f20f22

nlef, 2022-01-01:
9743669e45b44ec25e1aedc5fd7bed894f5dc34c adds saving lapse info into specific marco

aka13-404, 2022-02-04:
Done in v1.4

closed 2022-02-04 by aka13-404