nlef/moonraker-telegram-bot #66: Notification problem

Karol01pl, 2021-11-27:
For some time now, I have had a problem with notifications while printing, they stopped coming. Timelaps is also not working. The bot_restart command is not working.

[telegram.log](https://github.com/nlef/moonraker-telegram-bot/files/7612496/telegram.log)

Karol01pl, 2021-11-27:
[image]
Config file

nlef, 2021-11-28:
@Karol01pl  There are several problems with the log:
1.multiple bots launched with one token
2. strange names of time-lapse folders. most likely the folder name exceeds 64 characters (telegram limit)
Please provide the names of the time-lapse folders.
The temporary fix is to delete everything from the time-lapse folders. And name gcode files shorter than 64 characters

Karol01pl, 2021-11-28:
@nlef 
1. I only have one bot connected
2. Folder name: timelaps    /tmp/timelapse    Gcode name: test.gcode
timelapse folders are not created

nlef, 2021-11-28:
>     2. Folder name: timelaps    /tmp/timelapse    Gcode name: test.gcode
>        timelapse folders are not created
the screenshot shows a different path in the basedir parameter.
Please set cleanup paramter to false for debug reasons and check after that there is a folder and time-lapse images

Karol01pl, 2021-11-28:
I fixed it, it was enough to change the Token to another bot, reset the rpi and add the old Token back

closed 2021-11-28 by Karol01pl