nlef/moonraker-telegram-bot #73: High CPU usage causes mcu shutdown timer too close

ste94pz, 2021-12-19:
Hi, I'm using the latest version of fluiddpi with fluidd, moonraker and klipper all updated on a Raspberry Pi Zero 2. The printer is an Eryone Thinker S with an ATmega2560 based motherboard. I've had no problems until I installed the Telegram bot. During prints, I see high cpu usage by the bot process, it uses completely one of the four cores of the raspy, causing the print to fail after some time for timer errors with the mcu. In fluidd, I see an average of host cpu usage over 1.00, that falls rapidly around 0.20 if I stop the bot. In the bot's config, I enabled the notifications during the print only every 25% of completion and I included the temperatures of hotend and bed, no camera, timelapse or other functions. Is the high cpu usage normal? I'm here to help to diagnose the problem, if you need. Thank you!

1n9i9c7om, 2022-02-10:
I've had the same issue when I tested this bot last weekend on a Raspberry Pi 3B - I have therefore resorted to using the other moonraker telegram bot, but I like this one better. 
I had the timelapse feature enabled - should a Pi 3 be capable of running timelapse recording while also running Klipper (& Fluidd)?

ste94pz, 2022-02-10:
> It depends. Normally, if you are printing and simply take pictures for the lapse, the cpu load should be negligible. 
> 
> Now, BUILDING the lapse takes a lot of ressources. 
> 
> Maybe your system is already on the brink of failing, and the bot is simply "the last drop"? How do the loads look for you without the bot?

The load without the bot is very very low, I never had this problem before.

adawalli, 2022-02-11:
Glad this issue isn't just me - the moment I enabled this plugin, every single print fails with the timer error (thankfully only at the very end or very beginning). I have a raspberry pi3b, and don't even yet have my webcam plugged in (although I did configure it with fluidd and tested to make sure it worked)

ste94pz, 2022-02-11:
> Glad this issue isn't just me - the moment I enabled this plugin, every single print fails with the timer error (thankfully only at the very end or very beginning). I have a raspberry pi3b, and [...]

Do you have this problem even with the latest bot update? I haven't tried yet if it's solved now.

adawalli, 2022-02-11:
I installed for the first time last night (fresh install), but if something was released since then, I have not

ste94pz, 2022-02-11:
> I installed for the first time last night (fresh install), but if something was released since then, I have not

So the issue isn't solved in the latest version 😅

nlef, 2022-02-11:
> Glad this issue isn't just me - the moment I enabled this plugin, every single print fails with the timer error (thankfully only at the very end or very beginning). I have a raspberry pi3b, and [...]

Could you upload bot, klippy, moonraker logs? 
Bot is being tested on orangepi zero2, raspberypi 4 (2gb), raspberypi 3b+ (with cam & without). I'm confused and can't figure out why you're having problems

aka13-404, 2022-02-11:
> > I installed for the first time last night (fresh install), but if something was released since then, I have not
> 
> So the issue isn't solved in the latest version 😅

The thing is, we can't reproduce it, and by now we have about 1000 people using the bot, +-. Let's assume we have 10 times more people than in this ticket having the issues, that would be like 30 peope, or 3%. 

I don't understand, what exactly eats the memory in your setup. Is it the bot itself?

ste94pz, 2022-02-12:
> The thing is, we can't reproduce it, and by now we have about 1000 people using the bot, +-. Let's assume we have 10 times more people than in this ticket having the issues, that would be like 30 [...]

Please, don't get it wrong, I wasn't complaining. Anyway, I'll gather some infos to investigate the issue, it’s been a while since the last time that I used the printer. Thank you for the interest 💪✌️

adawalli, 2022-02-13:
> 

ACK - let me re-enable the plugin and I will get you the info you asked for if the issue still exists

adawalli, 2022-02-17:
I no longer have this issue - I think it may have been related to having a camera configured through my klipper ecosystem but not actually plugged in

aSocialMenace, 2023-03-02:
hey.. anyone figure this out? cuz i am having the same issue! telegram bot is using 65% of cpu resources on my pi zero 2 w causing MCU error timer to close. Installed the mainsail 32bit image using raspberry pi imager and then installed kiauh to update klipper, moonraker and mainsail then installed telegram bot. every time a print starts it fails druing the wipe line... 

printing on a ender 3 pro using a btt skr mini e3 v3 
pi is connected to the board using uart by way of the tft header