nlef/moonraker-telegram-bot #75: Allow for output pin configured light devices

th3fallen, 2021-12-24:
As you see here my chamber lights are controlled by an output pin
https://github.com/th3fallen/voronConfig/blob/master/output_pins.cfg#L2

but your config only allows for power devices, is there an option im missing for us non [power] plebs?

aka13-404, 2022-01-05:
The light functionality is very old, one of the very first features.
The idea initially behind having light only as a moonraker device was based on two ideas:

1) The bot can not check the status of the pin, if it is a klipper, and not a moonraker pin. So the bot has no idea, if the light is on, and if it should toggle it, or leave it be. (for example, if you have enabled the light manually, and want it to stay on for a while, you as a user would be propably dissatisfied, if the bot did disable it after every picture it takes. )
2) If you have seen it, there is a way to specify a delay between turning on the light, and taking the picture/video. Most cams need about 1-3 seconds to adjust to the changed lightning conditions. If the light device is sitting on the klipper side of things, there is a queue of commands, where pin changes get added to. Which means - we can't know for sure, how many seconds it will take to enable the light, and when it has been enabled, which may lead to "black" pictures.

Moonraker has changed a bit since then, and we do believe in user responsibility more since that time, and a lot of users which can not use moonraker light would be propably grateful, if the possibility would exist to toggle light via klipper pins.

We have a big update coming, and after it's done we will look into light devices. No promises on an ETA - propably somewhen at the end of february. 

I'll update this issue, as soon as I have more information.

nlef, 2022-02-04:
https://github.com/Arksine/moonraker/commit/0b13e2da44bdd0f806165f84a81282643ac24b7e

aka13-404, 2022-02-07:
@th3fallen With latest moonraker update you can now declare an out put pin from klipper to be a power device in moonraker.

I have a led strip defined in klipper as 
```
[output_pin chamber_led]
pin: PB6
```

And a moonraker device defined as 
```
[power chamber_led]
type: klipper_device
object_name: output_pin chamber_led
```

This works very well, you might want to try it out see if it works for you as well. 
I am closing this for now, but don't hesitate to reopen if it does not work as expected.

closed 2022-02-07 by aka13-404

th3fallen, 2022-02-07:
Thanks for the follow up I meant to close this when you linked to the moonraker commit. It works great. Thanks