nlef/moonraker-telegram-bot #77: Multiple moonraker instances on single machine

Glenn-Kerselaers, 2021-12-30:
I have installed two printer instances using KIAUH. I would like to receive notifications from both printers using moonraker-telegram-bot. The instructions for this situation aren't clear to me.
Do I need to install two separate moonraker-telegram-bot instances as well? The only thing i find about multiple instances is the following:

[bot]
server: localhost
#	This is the adress, where the moonraker of the desired printer is located at. 
#	In most cases it will be 'localhost'. Alternatively, an ip:port, as in 192.168.0.19:7125 can be entered, 
#	if you are running multiple moonraker instances on the machine, or if the bot is located not on the printer itself

Can someone explain to me what I need to do, please?

aka13-404, 2022-01-05:
Yeah, you are spot on. If you have two printers, (two moonrakers, two klippers, etc) you will also need two bots. They can share the same camera on most setups though. 

I have not tracked the progress with KIAUH, if it already can install two instances already.

tayanov, 2022-01-09:
another repo with moonraker telegramm bot have script to install two or more instance of bot. but bot not have buttons at menu telegram. https://github.com/Raabi91/moonraker-telegram

closed 2022-01-09 by aka13-404