nlef/moonraker-telegram-bot #79: ETA does not update

Krakert, 2022-01-05:
I was printing today with the bot in place and I noticed that the `Finish at` time does not update.
Mainsail ETA is lower than the bot reports. 

[image]

Running the most simple config
```
[progress_notification]
percent: 100
```

nlef, 2022-01-05:
We'll try to reproduce this and try to find the reason.

Krakert, 2022-01-06:
It does update, but it does not come close to the ETA that Moonraker shows? Maybe it is based on the file? I know that there are more estimated "done" times.

[image]
Time went from 00:12 to 00:13

aka13-404, 2022-01-10:
https://github.com/Klipper3d/klipper/blob/master/docs/Status_Reference.md#print_stats

Might be related M73/Displaystatus

aka13-404, 2022-02-04:
@Krakert Would you kindly check if this persists in 1.4?

Krakert, 2022-02-04:
Right now I'm setting up a new printer, when that's done I will setup the bot, a bit busy lately, but I will get to this! 

Great work guys!

aka13-404, 2022-02-05:
Don't hesitate to reopen if the issue persists.

closed 2022-02-05 by aka13-404