nlef/moonraker-telegram-bot #80: Option to config /status response

Krakert, 2022-01-06:
When you do `/status` you get the following output:

[image]

It would be nice to define that layout in the `telegram.conf`. Personally, I don't care about the filament usage and weight of the part (already checked in the slicer). I'm more interested in the layer height of the part, and maybe the total print time.

aka13-404, 2022-01-07:
I myself would like that as well. As you understand, the needs of all the different users are different, so there propably has to be some templating, a la Jinja2, and that is quite the undertaking. We are looking into it, but don't expect it in the near future.

Krakert, 2022-01-07:
Understandable! I really like the bot so far, really easy to set up!
Maybe I look into it myself in my free time, I like to contribute to the project. I already did some digging in the source code. 

Keep up the good work!

aka13-404, 2022-01-07:
We would love seeing contributions from you. Glad you enjoy it!

aka13-404, 2022-02-04:
Done in v1.4

closed 2022-02-04 by aka13-404