nlef/moonraker-telegram-bot #85: No timelapse at the end

INeDiA, 2022-01-17:
I had my timelapses working perfectly but they suddenly stopped working.
I get pictures and video updates without any issue, but at the end of the print I just read "Starting time-lapse assembly for..." and nothing happens.
I can't see any error in [telegram.log](https://github.com/nlef/moonraker-telegram-bot/files/7884008/telegram.log)
I have plenty of space on disk and nothing pops up on fluidd/moonraker at the end of the print.

I there any way to reset this behaviour? Tried to uninstall/install the whole thing but without success.
Thanks!

nlef, 2022-01-17:
Judging by the log, you have problems connecting to telegram servers.
After restarting, the bot should send messages with a proposal to rebuild unbuided timelapses.

INeDiA, 2022-01-17:
This is weird, it's been a few days now and I of course had restarted my printer...
It says "printer online" when I start it up and I can send commands to the bot; the only thing that does not work is the timelapse.

pidpawel, 2022-02-03:
I believe I'm having the same issue. All the regular messages are fine, but the timelapses almost never finish. The error in the logs is `telegram.vendor.ptb_urllib3.urllib3.exceptions.ReadTimeoutError: HTTPSConnectionPool(host='api.telegram.org', port=443): Read timed out. (read timeout=5.0)` As the "timelapse progress bar" is sending messages kinda quickly I think telegram API is temporarily rejecting those messages as a kind of spam prevention. The solution may be sending updates every n-seconds instead.

pidpawel, 2022-02-03:
Yup, if I comment out lines for `./bot/camera.py:                    info_mess.edit_text(text=f"Images recoded {fnum}/{photo_count}")` it seems to be working just fine.

closed 2022-02-04 by aka13-404

INeDiA, 2022-02-05:
> Thanks to the fix from @pidpawel this should no longer be an issue. Don't hesitate to reopen if it persists.

I'm sorry, is this updated in v1.4-9? Because I still get the same "error".
I still can't see any error in logs :(
This is so weird!