nlef/moonraker-telegram-bot #97: [Bug]: Timelapse broken on latest build

ssstraub, 2022-02-04:
### What happened and in what context?

I had telegram bot working fine on the prior release. Just updated today to the latest release and now it:

- Never shows a video preview of any kind
- Fails to generate timelapses 
- Still responds normally to commands in my telegram bot channel.

So basically all video-related functions seem to have broken. 

here's telegram.log:
```
2022-02-04 16:16:56,473 - __main__ - ERROR - Exception while handling an update:
Traceback (most recent call last):
  File "/home/pi/moonraker-telegram-bot-env/lib/python3.7/site-packages/telegram/ext/dispatcher.py", line 555, in process_update
    handler.handle_update(update, self, check, context)
  File "/home/pi/moonraker-telegram-bot-env/lib/python3.7/site-packages/telegram/ext/handler.py", line 198, in handle_update
    return self.callback(update, context)
  File "/home/pi/moonraker-telegram-bot/bot/main.py", line 327, in button_handler
    (video_bio, thumb_bio, width, height, video_path, gcode_name) = cameraWrap.create_timelapse_for_file(lapse_name, info_mess)
  File "/home/pi/moonraker-telegram-bot/bot/camera.py", line 376, in create_timelapse_for_file
    return self._create_timelapse(filename, filename, info_mess)
  File "/home/pi/moonraker-telegram-bot/bot/camera.py", line 405, in _create_timelapse
    last_photo = photos[-1]
IndexError: list index out of range
2022-02-04 16:18:29,894 - __main__ - ERROR - Exception while handling an update:
Traceback (most recent call last):
  File "/home/pi/moonraker-telegram-bot-env/lib/python3.7/site-packages/PIL/JpegImagePlugin.py", line 630, in _save
    rawmode = RAWMODE[im.mode]
KeyError: 'P'

The above exception was the direct cause of the following exception:

Traceback (most recent call last):
  File "/home/pi/moonraker-telegram-bot-env/lib/python3.7/site-packages/telegram/ext/utils/promise.py", line 96, in run
    self._result = self.pooled_function(*self.args, **self.kwargs)
  File "/home/pi/moonraker-telegram-bot/bot/main.py", line 111, in status
    with cameraWrap.take_photo() as bio:
  File "/home/pi/moonraker-telegram-bot/bot/camera.py", line 43, in wrapper
    result = func(self, *args, **kwargs)
  File "/home/pi/moonraker-telegram-bot/bot/camera.py", line 255, in take_photo
    img.save(bio, 'JPEG', quality=80, subsampling=0)
  File "/home/pi/moonraker-telegram-bot-env/lib/python3.7/site-packages/PIL/Image.py", line 2212, in save
    save_handler(self, fp, filename)
  File "/home/pi/moonraker-telegram-bot-env/lib/python3.7/site-packages/PIL/JpegImagePlugin.py", line 632, in _save
    raise OSError(f"cannot write mode {im.mode} as JPEG") from e
OSError: cannot write mode P as JPEG
2022-02-04 16:18:35,600 - __main__ - ERROR - Exception while handling an update:
Traceback (most recent call last):
  File "/home/pi/moonraker-telegram-bot-env/lib/python3.7/site-packages/PIL/JpegImagePlugin.py", line 630, in _save
    rawmode = RAWMODE[im.mode]
KeyError: 'P'
```

### What branch does this occur on?

master

### Reports only on latest master or development branches!

- [X] I have updated to latest development/master version before submitting the bug

ssstraub, 2022-02-04:
Here's a dump of my journalctl

```
-- Logs begin at Thu 2022-02-03 17:44:14 CST, end at Fri 2022-02-04 16:25:25 CST. --
Feb 04 16:21:15 TridentPi python[26535]: 2022-02-04 16:21:15,031 - apscheduler.scheduler - INFO - Removed job ws_reschedule
Feb 04 16:21:15 TridentPi python[26535]: 2022-02-04 16:21:15,008 - apscheduler.scheduler - INFO - Added job "reshedule" to job store "default"
Feb 04 16:21:12 TridentPi python[26535]: 2022-02-04 16:21:12,281 - apscheduler.scheduler - INFO - Scheduler started
Feb 04 16:21:11 TridentPi python[26535]: 2022-02-04 16:21:11,024 - apscheduler.scheduler - INFO - Scheduler started
Feb 04 16:21:04 TridentPi systemd[1]: Started Starts Moonraker Telegram Bot on startup.
Feb 04 16:21:04 TridentPi systemd[1]: Stopped Starts Moonraker Telegram Bot on startup.
Feb 04 16:21:04 TridentPi systemd[1]: moonraker-telegram-bot.service: Scheduled restart job, restart counter is at 7.
Feb 04 16:21:04 TridentPi systemd[1]: moonraker-telegram-bot.service: Service RestartSec=5s expired, scheduling restart.
Feb 04 16:20:59 TridentPi systemd[1]: moonraker-telegram-bot.service: Failed with result 'exit-code'.
Feb 04 16:20:59 TridentPi systemd[1]: moonraker-telegram-bot.service: Main process exited, code=exited, status=1/FAILURE
Feb 04 16:20:59 TridentPi python[23326]: 2022-02-04 16:20:59,370 - apscheduler.scheduler - INFO - Scheduler has been shut down
Feb 04 16:20:59 TridentPi python[23326]: 2022-02-04 16:20:59,369 - __main__ - INFO - Exiting! Moonraker connection lost!
Feb 04 16:20:59 TridentPi python[23326]: 2022-02-04 16:20:59,367 - __main__ - INFO - WebSocket closed
Feb 04 16:18:43 TridentPi python[23326]: OSError: cannot write mode P as JPEG
Feb 04 16:18:43 TridentPi python[23326]:     raise OSError(f"cannot write mode {im.mode} as JPEG") from e
Feb 04 16:18:43 TridentPi python[23326]:   File "/home/pi/moonraker-telegram-bot-env/lib/python3.7/site-packages/PIL/JpegImagePlugin.py", line 632, in _s
Feb 04 16:18:43 TridentPi python[23326]:     save_handler(self, fp, filename)
Feb 04 16:18:43 TridentPi python[23326]:   File "/home/pi/moonraker-telegram-bot-env/lib/python3.7/site-packages/PIL/Image.py", line 2212, in save
Feb 04 16:18:43 TridentPi python[23326]:     img.save(bio, 'JPEG', quality=80, subsampling=0)
Feb 04 16:18:43 TridentPi python[23326]:   File "/home/pi/moonraker-telegram-bot/bot/camera.py", line 255, in take_photo
Feb 04 16:18:43 TridentPi python[23326]:     result = func(self, *args, **kwargs)
Feb 04 16:18:43 TridentPi python[23326]:   File "/home/pi/moonraker-telegram-bot/bot/camera.py", line 43, in wrapper
Feb 04 16:18:43 TridentPi python[23326]:     with cameraWrap.take_photo() as bio:
Feb 04 16:18:43 TridentPi python[23326]:   File "/home/pi/moonraker-telegram-bot/bot/main.py", line 111, in status
Feb 04 16:18:43 TridentPi python[23326]:     self._result = self.pooled_function(*self.args, **self.kwargs)
Feb 04 16:18:43 TridentPi python[23326]:   File "/home/pi/moonraker-telegram-bot-env/lib/python3.7/site-packages/telegram/ext/utils/promise.py", line 96,
Feb 04 16:18:43 TridentPi python[23326]: Traceback (most recent call last):
Feb 04 16:18:43 TridentPi python[23326]: The above exception was the direct cause of the following exception:
Feb 04 16:18:43 TridentPi python[23326]: KeyError: 'P'
Feb 04 16:18:43 TridentPi python[23326]:     rawmode = RAWMODE[im.mode]
Feb 04 16:18:43 TridentPi python[23326]:   File "/home/pi/moonraker-telegram-bot-env/lib/python3.7/site-packages/PIL/JpegImagePlugin.py", line 630, in _s
Feb 04 16:18:43 TridentPi python[23326]: Traceback (most recent call last):
Feb 04 16:18:43 TridentPi python[23326]: 2022-02-04 16:18:43,404 - __main__ - ERROR - Exception while handling an update:
Feb 04 16:18:35 TridentPi python[23326]: OSError: cannot write mode P as JPEG
Feb 04 16:18:35 TridentPi python[23326]:     raise OSError(f"cannot write mode {im.mode} as JPEG") from e
Feb 04 16:18:35 TridentPi python[23326]:   File "/home/pi/moonraker-telegram-bot-env/lib/python3.7/site-packages/PIL/JpegImagePlugin.py", line 632, in _s
Feb 04 16:18:35 TridentPi python[23326]:     save_handler(self, fp, filename)
Feb 04 16:18:35 TridentPi python[23326]:   File "/home/pi/moonraker-telegram-bot-env/lib/python3.7/site-packages/PIL/Image.py", line 2212, in save
Feb 04 16:18:35 TridentPi python[23326]:     img.save(bio, 'JPEG', quality=80, subsampling=0)
Feb 04 16:18:35 TridentPi python[23326]:   File "/home/pi/moonraker-telegram-bot/bot/camera.py", line 255, in take_photo
Feb 04 16:18:35 TridentPi python[23326]:     result = func(self, *args, **kwargs)
Feb 04 16:18:35 TridentPi python[23326]:   File "/home/pi/moonraker-telegram-bot/bot/camera.py", line 43, in wrapper
```

nlef, 2022-02-05:
@ssstraub Does the status button work? Does bot send photos? Is the url to the stream from the camera specified in the configuration?

drveni, 2022-02-05:
Same here.. same problems after last update... Same errors. No timelapse, no pic after 5% of print,... Etc...

drveni, 2022-02-05:
> @ssstraub Does the status button work? Does bot send photos? Is the url to the stream from the camera specified in the configuration?

My last config works fine.. after update not work.

I'm not have link to camera in my config not need to have it works.. now I need to put video stream .. and it start show when I put /photo but no more start photo not more timelapse.. not more photo at every 5% print... or 1800sec print..

On telegram chat changed just /video where /picture link gone?? Whay? If you add more controls on left button whay you changed most usage stuff on right shortcut button!? so why that?

On after print still won't sent that print is finished and picture..!? 

On telegram message printer
Printer online
Unknown/bad items in [camera] section:
    flipvertically: false
    fliphorizontally: false
    videoduration: 15

Where flip status gone? Now default video duration is 5 sec!? Not 15 as default..!?

nlef, 2022-02-05:
@ssstraub please, take a look at release notes for v1.4. everything is described...

drveni, 2022-02-05:
> @ssstraub please, take a look at release notes for v1.4. everything is described...

fixed.. but still not working pic send on percent, and send finished timelapse.... and defined time: like timer.

on command /picture /video now work

drveni, 2022-02-05:
fixed.. but still not working pic send on percent, and send finished timelapse.... and defined time: like timer.

on command /picture /video now work

ssstraub, 2022-02-05:
> @ssstraub Does the status button work? Does bot send photos? Is the url to the stream from the camera specified in the configuration?

The Status button does NOT work since updating. No photos or videos of any kind function.

ssstraub, 2022-02-05:
From what I can see, the specific error that is breaking all webcam functions is `cannot write mode P as JPEG`

drveni, 2022-02-06:
I found another interesting problem. not to open a new topic, I will continue here because it has to do with this.

namely on the telegram mobile application. chat with my bot. it is always one and the same message. as if the old message was deleted and a new one appeared with new data. I don't know if you understand. namely, the image and status of the print. i turn on the same chat again be the same last message one only with new details from the print. so a new message does not come to be chronological, but to copy over the same message. I will send pictures in the attachment. I hope they understand you. new messages do not come only the same every time ... as if the message has the same data (time offset), but the content is different. 

[image]
[image]

aka13-404, 2022-02-06:
> I found another interesting problem. not to open a new topic, I will continue here because it has to do with this.
>
> namely on the telegram mobile application. chat with my bot. it is always one and [...]
  

This is the idea behind single-message notifications, you don't get spammed with notifications, but instead 1 message per print is sent and then updated. Not sure I understand what you mean?

drveni, 2022-02-06:
> This is the idea behind single-message notifications, you don't get spammed with notifications, but instead 1 message per print is sent and then updated. Not sure I understand what you mean?

so then new messages do not come to the cell phone for example 5% that the printer has printed. if it's already an idea, then leave this and that old option for messages to come. isn't there in the settings already made not to spam with "min_delay_between_notifications: 60"?

however, notifications from the printer do not come at all if it is set to 5% to send the image. and no timelapse to send on mobile. ?! it is lit in the settings. maybe I'm doing something wrong. but after the last update it doesn't work

aka13-404, 2022-02-06:
Please try to read the changelog, before posting, thank you.

closed 2022-02-06 by aka13-404

ssstraub, 2022-02-06:
@nlef 

> Is the url to the stream from the camera specified in the configuration?

This was the problem... Host URL wasn't necessary before but is in the current release. Just noting here in case anyone else has this same issue.