nlef/moonraker-telegram-bot #104: [Bug]: After update not working anymore

Technofrikus, 2022-02-12:
### What happened and in what context?

The telegram bot was installed and worked fine. telegram.conf is in the same directory as printer.cfg

Then I updated a few days ago to the newest version (first update of the bot ever) and it did not work anymore. The Bot is running according to the menu in the power button in mainsail but there are no messages from the bot and it doesnt answer any in Telegram.

I then used Kiauh to uninstall and reinstall the bot. I tried several times, but the same problem. I also redid the installation by hand. Nothing.

During installation with Kiauh it says:
```
###### Creating base config file
cp: cannot create regular file '/telegram.conf': Permission denied

>>>>>> MoonrakerTelegramBot successfully installed!

###### Restarting MoonrakerTelegramBot Service ...
Failed to restart moonraker-telegram-bot.service: Unit moonraker-telegram-bot.service not found.
```

During manual installation there are no error messages.

The telegram.conf is still there and untouched. 

There are also no logs anymore. I read a log of the bot once, but there is no telegram log file anymore in klipper_logs, although I left the path untouched during installation.

### If needed, paste the relevant bot journal contents here.

-- Logs begin at Thu 2019-02-14 10:11:59 GMT, end at Sat 2022-02-12 14:24:06 GMT. --
Feb 12 14:24:02 ratos systemd[1]: Started Starts Moonraker Telegram Bot on startup.
Feb 12 14:24:02 ratos systemd[1]: Stopped Starts Moonraker Telegram Bot on startup.
Feb 12 14:24:02 ratos systemd[1]: moonraker-telegram-bot.service: Scheduled restart job, restart counter is at 27.
Feb 12 14:24:02 ratos systemd[1]: moonraker-telegram-bot.service: Service RestartSec=5s expired, scheduling restart.
Feb 12 14:23:56 ratos systemd[1]: moonraker-telegram-bot.service: Failed with result 'exit-code'.
Feb 12 14:23:56 ratos systemd[1]: moonraker-telegram-bot.service: Main process exited, code=exited, status=1/FAILURE
Feb 12 14:23:55 ratos python[3300]:     logger.error("Uncaught exception", exc_info=(exc_type, exc_value, exc_traceback), stack_info=True)
Feb 12 14:23:55 ratos python[3300]:   File "/home/pi/moonraker-telegram-bot/bot/main.py", line 55, in handle_exception
Feb 12 14:23:55 ratos python[3300]: Stack (most recent call last):
Feb 12 14:23:55 ratos python[3300]: configparser.NoOptionError: No option 'host' in section: 'camera'
Feb 12 14:23:55 ratos python[3300]:     raise NoOptionError(option, section)
Feb 12 14:23:55 ratos python[3300]:   File "/usr/lib/python3.7/configparser.py", line 791, in get
Feb 12 14:23:55 ratos python[3300]:     self.host: str = config.get(self._SECTION, 'host')
Feb 12 14:23:55 ratos python[3300]:   File "/home/pi/moonraker-telegram-bot/bot/configuration.py", line 44, in __init__
Feb 12 14:23:55 ratos python[3300]:     self.camera = CameraConfig(config)
Feb 12 14:23:55 ratos python[3300]:   File "/home/pi/moonraker-telegram-bot/bot/configuration.py", line 129, in __init__
Feb 12 14:23:55 ratos python[3300]:     configWrap = ConfigWrapper(conf)
Feb 12 14:23:55 ratos python[3300]:   File "/home/pi/moonraker-telegram-bot/bot/main.py", line 893, in <module>
Feb 12 14:23:55 ratos python[3300]: Traceback (most recent call last):
Feb 12 14:23:55 ratos python[3300]: During handling of the above exception, another exception occurred:
Feb 12 14:23:55 ratos python[3300]: KeyError: 'host'
Feb 12 14:23:55 ratos python[3300]:     raise KeyError(key)
Feb 12 14:23:55 ratos python[3300]:   File "/home/pi/moonraker-telegram-bot-env/lib/python3.7/collections/__init__.py", line 906, in __missing__
Feb 12 14:23:55 ratos python[3300]:     return self.__missing__(key)            # support subclasses that define __missing__
Feb 12 14:23:55 ratos python[3300]:   File "/home/pi/moonraker-telegram-bot-env/lib/python3.7/collections/__init__.py", line 914, in __getitem__
Feb 12 14:23:55 ratos python[3300]:     value = d[option]
Feb 12 14:23:55 ratos python[3300]:   File "/usr/lib/python3.7/configparser.py", line 788, in get
Feb 12 14:23:55 ratos python[3300]: Traceback (most recent call last):
Feb 12 14:23:55 ratos python[3300]: 2022-02-12 14:23:55,482 - __main__ - ERROR - Uncaught exception
Feb 12 14:23:48 ratos systemd[1]: Started Starts Moonraker Telegram Bot on startup.
Feb 12 14:23:48 ratos systemd[1]: Stopped Starts Moonraker Telegram Bot on startup.
Feb 12 14:23:48 ratos systemd[1]: moonraker-telegram-bot.service: Scheduled restart job, restart counter is at 26.
Feb 12 14:23:48 ratos systemd[1]: moonraker-telegram-bot.service: Service RestartSec=5s expired, scheduling restart.
Feb 12 14:23:43 ratos systemd[1]: moonraker-telegram-bot.service: Failed with result 'exit-code'.
Feb 12 14:23:43 ratos systemd[1]: moonraker-telegram-bot.service: Main process exited, code=exited, status=1/FAILURE
Feb 12 14:23:42 ratos python[3282]:     logger.error("Uncaught exception", exc_info=(exc_type, exc_value, exc_traceback), stack_info=True)
Feb 12 14:23:42 ratos python[3282]:   File "/home/pi/moonraker-telegram-bot/bot/main.py", line 55, in handle_exception
Feb 12 14:23:42 ratos python[3282]: Stack (most recent call last):
Feb 12 14:23:42 ratos python[3282]: configparser.NoOptionError: No option 'host' in section: 'camera'
Feb 12 14:23:42 ratos python[3282]:     raise NoOptionError(option, section)
Feb 12 14:23:42 ratos python[3282]:   File "/usr/lib/python3.7/configparser.py", line 791, in get
Feb 12 14:23:42 ratos python[3282]:     self.host: str = config.get(self._SECTION, 'host')
Feb 12 14:23:42 ratos python[3282]:   File "/home/pi/moonraker-telegram-bot/bot/configuration.py", line 44, in __init__
Feb 12 14:23:42 ratos python[3282]:     self.camera = CameraConfig(config)
Feb 12 14:23:42 ratos python[3282]:   File "/home/pi/moonraker-telegram-bot/bot/configuration.py", line 129, in __init__
Feb 12 14:23:42 ratos python[3282]:     configWrap = ConfigWrapper(conf)
Feb 12 14:23:42 ratos python[3282]:   File "/home/pi/moonraker-telegram-bot/bot/main.py", line 893, in <module>
Feb 12 14:23:42 ratos python[3282]: Traceback (most recent call last):
Feb 12 14:23:42 ratos python[3282]: During handling of the above exception, another exception occurred:
Feb 12 14:23:42 ratos python[3282]: KeyError: 'host'
Feb 12 14:23:42 ratos python[3282]:     raise KeyError(key)
Feb 12 14:23:42 ratos python[3282]:   File "/home/pi/moonraker-telegram-bot-env/lib/python3.7/collections/__init__.py", line 906, in __missing__
Feb 12 14:23:42 ratos python[3282]:     return self.__missing__(key)            # support subclasses that define __missing__
Feb 12 14:23:42 ratos python[3282]:   File "/home/pi/moonraker-telegram-bot-env/lib/python3.7/collections/__init__.py", line 914, in __getitem__
Feb 12 14:23:42 ratos python[3282]:     value = d[option]
Feb 12 14:23:42 ratos python[3282]:   File "/usr/lib/python3.7/configparser.py", line 788, in get

### What branch does this occur on?

master

### Reports only on latest master or development branches!

- [X] I have updated to latest development/master version before submitting the bug

aka13-404, 2022-02-12:
You did not provide the config, but I assume based on the log, that you have not defined a camera, but want to use timelapse or similar camera related functionality. 

If you want to do picture stuff, you gotta say where your camera/images are coming from.

Technofrikus, 2022-02-12:
That indeed solved it. I added the camera information and now it works. I dont have a camera installed, so I did not add it and it wasnt a problem before. Just saw a nice Nyan cat :) 

Some options were renamed it seems:
```
Printer online
Unknown/bad items in [camera] section:
    flipvertically: false
    fliphorizontally: false
    videoduration: 15

Unknown/bad items in [progress_notification] section:
    min_delay_between_notifications: 60
```

but that is not a problem. I will update them to the new names. 

Thank you very much!

aka13-404, 2022-02-12:
> That indeed solved it. I added the camera information and now it works. I dont have a camera installed, so I did not add it and it wasnt a problem before. Just saw a nice Nyan cat :)
>
> Some options [...]

This changes everything! 
Would you be so kind to paste the config that you were running, before you added the camera?
The bot should work without the camera, if you do not try to configure it later, the "no-camera-mode" as we call it.
So I am wondering, what you had configured, that it "wanted" to see it configured.

Technofrikus, 2022-02-12:
```
[bot]
server: localhost
chat_id: xxx
bot_token: xxx
log_path: /home/pi/klipper_logs

[camera]
flipVertically: false
flipHorizontally: false
videoDuration: 15

[progress_notification]
percent: 50
height: 0
time: 0
min_delay_between_notifications: 60

[timelapse]
cleanup: true
height: 0.2
time: 5
target_fps: 30	

```

Sure, no problem. This was the config.

New config:
```
[bot]
server: localhost
chat_id: xxx
bot_token: xxx
log_path: /home/pi/klipper_logs

[camera]
host: http://localhost:8080/?action=stream

[progress_notification]
percent: 50
height: 0
time: 0
```

I also tried to remove the camera section but it is not optional it seems. Not sure how to activate no-camera-mode? No it just always says "no signal".

I started a print now and the bot is restarting every minute or so. This is the bot journal:

```
-- Logs begin at Sat 2022-02-12 15:18:27 GMT, end at Sat 2022-02-12 15:55:59 GMT. --
Feb 12 15:55:50 ratos python[5123]: 2022-02-12 15:55:50,310 - apscheduler.scheduler - INFO - Removed job ws_reschedule
Feb 12 15:55:50 ratos python[5123]: 2022-02-12 15:55:50,290 - apscheduler.scheduler - INFO - Added job "reshedule" to job store "default"
Feb 12 15:55:49 ratos python[5123]: 2022-02-12 15:55:49,538 - apscheduler.scheduler - INFO - Scheduler started
Feb 12 15:55:49 ratos python[5123]: 2022-02-12 15:55:49,368 - apscheduler.scheduler - INFO - Scheduler started
Feb 12 15:55:43 ratos systemd[1]: Started Starts Moonraker Telegram Bot on startup.
Feb 12 15:55:43 ratos systemd[1]: Stopped Starts Moonraker Telegram Bot on startup.
Feb 12 15:55:43 ratos systemd[1]: moonraker-telegram-bot.service: Scheduled restart job, restart counter is at 10.
Feb 12 15:55:43 ratos systemd[1]: moonraker-telegram-bot.service: Service RestartSec=5s expired, scheduling restart.
Feb 12 15:55:38 ratos systemd[1]: moonraker-telegram-bot.service: Succeeded.
Feb 12 15:55:30 ratos python[5036]: 2022-02-12 15:55:30,231 - apscheduler.scheduler - INFO - Scheduler has been shut down
Feb 12 15:55:30 ratos python[5036]: 2022-02-12 15:55:30,230 - __main__ - INFO - Exiting! Moonraker connection lost!
Feb 12 15:55:30 ratos python[5036]: 2022-02-12 15:55:30,229 - __main__ - INFO - WebSocket closed
Feb 12 15:54:57 ratos python[5036]: 2022-02-12 15:54:57,404 - apscheduler.scheduler - INFO - Removed job ws_reschedule
Feb 12 15:54:57 ratos python[5036]: 2022-02-12 15:54:57,379 - apscheduler.scheduler - INFO - Added job "reshedule" to job store "default"
Feb 12 15:54:56 ratos python[5036]: 2022-02-12 15:54:56,496 - apscheduler.scheduler - INFO - Scheduler started
Feb 12 15:54:56 ratos python[5036]: 2022-02-12 15:54:56,267 - apscheduler.scheduler - INFO - Scheduler started
Feb 12 15:54:49 ratos systemd[1]: Started Starts Moonraker Telegram Bot on startup.
Feb 12 15:54:49 ratos systemd[1]: Stopped Starts Moonraker Telegram Bot on startup.
Feb 12 15:54:49 ratos systemd[1]: moonraker-telegram-bot.service: Scheduled restart job, restart counter is at 9.
[... 5 lines cut ...]
Feb 12 15:54:03 ratos python[4947]: 2022-02-12 15:54:03,907 - apscheduler.scheduler - INFO - Removed job ws_reschedule
Feb 12 15:54:03 ratos python[4947]: 2022-02-12 15:54:03,885 - apscheduler.scheduler - INFO - Added job "reshedule" to job store "default"
Feb 12 15:54:02 ratos python[4947]: 2022-02-12 15:54:02,946 - apscheduler.scheduler - INFO - Scheduler started
Feb 12 15:54:02 ratos python[4947]: 2022-02-12 15:54:02,777 - apscheduler.scheduler - INFO - Scheduler started
Feb 12 15:53:56 ratos systemd[1]: Started Starts Moonraker Telegram Bot on startup.
Feb 12 15:53:56 ratos systemd[1]: Stopped Starts Moonraker Telegram Bot on startup.
Feb 12 15:53:56 ratos systemd[1]: moonraker-telegram-bot.service: Scheduled restart job, restart counter is at 8.
Feb 12 15:53:56 ratos systemd[1]: moonraker-telegram-bot.service: Service RestartSec=5s expired, scheduling restart.
Feb 12 15:53:51 ratos systemd[1]: moonraker-telegram-bot.service: Succeeded.
Feb 12 15:53:43 ratos python[4858]: 2022-02-12 15:53:43,172 - apscheduler.scheduler - INFO - Scheduler has been shut down
Feb 12 15:53:43 ratos python[4858]: 2022-02-12 15:53:43,171 - __main__ - INFO - Exiting! Moonraker connection lost!
Feb 12 15:53:43 ratos python[4858]: 2022-02-12 15:53:43,168 - __main__ - INFO - WebSocket closed
Feb 12 15:53:10 ratos python[4858]: 2022-02-12 15:53:10,522 - apscheduler.scheduler - INFO - Removed job ws_reschedule
Feb 12 15:53:10 ratos python[4858]: 2022-02-12 15:53:10,495 - apscheduler.scheduler - INFO - Added job "reshedule" to job store "default"
Feb 12 15:53:09 ratos python[4858]: 2022-02-12 15:53:09,630 - apscheduler.scheduler - INFO - Scheduler started
Feb 12 15:53:09 ratos python[4858]: 2022-02-12 15:53:09,456 - apscheduler.scheduler - INFO - Scheduler started
Feb 12 15:53:03 ratos systemd[1]: Started Starts Moonraker Telegram Bot on startup.
Feb 12 15:53:03 ratos systemd[1]: Stopped Starts Moonraker Telegram Bot on startup.
Feb 12 15:53:03 ratos systemd[1]: moonraker-telegram-bot.service: Scheduled restart job, restart counter is at 7.
Feb 12 15:53:03 ratos systemd[1]: moonraker-telegram-bot.service: Service RestartSec=5s expired, scheduling restart.
Feb 12 15:52:58 ratos systemd[1]: moonraker-telegram-bot.service: Succeeded.
Feb 12 15:52:49 ratos python[4774]: 2022-02-12 15:52:49,962 - apscheduler.scheduler - INFO - Scheduler has been shut down
Feb 12 15:52:49 ratos python[4774]: 2022-02-12 15:52:49,961 - __main__ - INFO - Exiting! Moonraker connection lost!
Feb 12 15:52:49 ratos python[4774]: 2022-02-12 15:52:49,960 - __main__ - INFO - WebSocket closed
```

nlef, 2022-02-12:
@Technofrikus 
please try the new version. I fixed a bug with host parsing when the camera section was disabled

Technofrikus, 2022-02-13:
I left the camera settings in there at first because I didnt think about it. It kind of worked, but it seemed the bot was waiting for something (probably the stream from the nonexisiting webcam) and was very slow to respond.

Then I removed the complete camera section and now it works great! No pictures, just written updates. Was a bit confused at first when hitting status that it was removed shortly after, but then understood, that the last message was updated. Very clever :)

Thanks for the help!

closed 2022-02-13 by nlef