nlef/moonraker-telegram-bot #116: [Bug]: A useful, descriptive title

crimsonclyde, 2022-02-20:
### What happened and in what context?

Updated Telegram Bot with KIAUH to v1.4.1.-14

Now command /photo is not working any more.

KIAUH v3.1.0-95
Moonraker: v0.7.1.-376

Config Camera
```
[camera]
host: http://localhost:8080/?action=stream
```

Removed and fresh installed Telegram-Bot again via KIAUH. Same issue.

### If needed, paste the relevant bot log contents here.

Empty.

### If needed, paste the relevant bot journal contents here.

Feb 18 20:15:36 fluiddpi python[3512]: 2022-02-18 20:15:36,831 - camera - WARNING - Reading video frames queue exception <built-in method with_traceback of Empty object at 0xa40643e8>
Feb 18 20:15:36 fluiddpi python[3512]: 2022-02-18 20:15:36,763 - camera - WARNING - Reading video frames queue exception <built-in method with_traceback of Empty object at 0xa40643e8>

### What branch does this occur on?

master

### Reports only on latest master or development branches!

- [X] I have updated to latest development/master version before submitting the bug

aka13-404, 2022-02-20:
This is not a bug. Please read release notes before updating.

closed 2022-02-20 by aka13-404

crimsonclyde, 2022-02-20:
Ahh thx - found it. For others:

/photo is removed. /status is now the replacement, and sends a picture together with configurable info on the printer.