nlef/moonraker-telegram-bot #117: [Bug]: mp4 file deleted if size exceed 50MB

CODeRUS, 2022-02-21:
### What happened and in what context?

i have `cleanup: true` in config
timelapse is bigger than 50MB
got message i should grab it from folder myself
but no mp4 here
but a lot of webp
imo cleanup could delete webp but preserve mp4 if not sent by any conditions

### What branch does this occur on?

development

### Reports only on latest master or development branches!

- [X] I have updated to latest development/master version before submitting the bug

nlef, 2022-02-23:
@CODeRUS pls check fix)

CODeRUS, 2022-02-26:
works fine

closed 2022-02-26 by CODeRUS