nlef/moonraker-telegram-bot #119: Bot online, no moonraker connection! error

teslabol77, 2022-02-25:
### What happened and in what context?

I followed the install instructions and I am not able to get this to work. I just keep getting a repeating text from the bot:

Bot online, no moonraker connection!
Connection failed, Unauthorized
Failing...

I am not sure how to do a clean install as there are no instructions for that. 

Thanks for the help. 

### If needed, paste the relevant bot log contents here.

2022-02-24 18:05:45,611 - __main__ - ERROR - Handshake status 401 Unauthorized
2022-02-24 18:05:45,611 - __main__ - INFO - WebSocket closed
2022-02-24 18:05:45,612 - __main__ - INFO - Exiting! Moonraker connection lost!
2022-02-24 18:06:05,729 - __main__ - ERROR - Handshake status 401 Unauthorized
2022-02-24 18:06:05,730 - __main__ - INFO - WebSocket closed
2022-02-24 18:06:05,730 - __main__ - INFO - Exiting! Moonraker connection lost!
2022-02-24 18:06:25,649 - __main__ - ERROR - Handshake status 401 Unauthorized
2022-02-24 18:06:25,650 - __main__ - INFO - WebSocket closed
2022-02-24 18:06:25,650 - __main__ - INFO - Exiting! Moonraker connection lost!
2022-02-24 18:06:45,591 - __main__ - ERROR - Handshake status 401 Unauthorized
2022-02-24 18:06:45,591 - __main__ - INFO - WebSocket closed
2022-02-24 18:06:45,592 - __main__ - INFO - Exiting! Moonraker connection lost!
2022-02-24 18:07:05,549 - __main__ - ERROR - Handshake status 401 Unauthorized
2022-02-24 18:07:05,549 - __main__ - INFO - WebSocket closed
2022-02-24 18:07:05,550 - __main__ - INFO - Exiting! Moonraker connection lost!
2022-02-24 18:07:25,413 - __main__ - ERROR - Handshake status 401 Unauthorized
2022-02-24 18:07:25,413 - __main__ - INFO - WebSocket closed
2022-02-24 18:07:25,414 - __main__ - INFO - Exiting! Moonraker connection lost!
2022-02-24 18:07:45,337 - __main__ - ERROR - Handshake status 401 Unauthorized
2022-02-24 18:07:45,338 - __main__ - INFO - WebSocket closed
2022-02-24 18:07:45,338 - __main__ - INFO - Exiting! Moonraker connection lost!
2022-02-24 18:08:05,435 - __main__ - ERROR - Handshake status 401 Unauthorized
2022-02-24 18:08:05,436 - __main__ - INFO - WebSocket closed
2022-02-24 18:08:05,437 - __main__ - INFO - Exiting! Moonraker connection lost!
2022-02-24 18:08:25,485 - __main__ - ERROR - Handshake status 401 Unauthorized
2022-02-24 18:08:25,486 - __main__ - INFO - WebSocket closed
2022-02-24 18:08:25,486 - __main__ - INFO - Exiting! Moonraker connection lost!
2022-02-24 18:08:45,404 - __main__ - ERROR - Handshake status 401 Unauthorized
2022-02-24 18:08:45,405 - __main__ - INFO - WebSocket closed
2022-02-24 18:08:45,406 - __main__ - INFO - Exiting! Moonraker connection lost!
2022-02-24 18:09:05,401 - __main__ - ERROR - Handshake status 401 Unauthorized
2022-02-24 18:09:05,401 - __main__ - INFO - WebSocket closed
2022-02-24 18:09:05,401 - __main__ - INFO - Exiting! Moonraker connection lost!
2022-02-24 18:09:25,421 - __main__ - ERROR - Handshake status 401 Unauthorized
2022-02-24 18:09:25,421 - __main__ - INFO - WebSocket closed
2022-02-24 18:09:25,422 - __main__ - INFO - Exiting! Moonraker connection lost!
2022-02-24 18:09:45,422 - __main__ - ERROR - Handshake status 401 Unauthorized
2022-02-24 18:09:45,423 - __main__ - INFO - WebSocket closed
2022-02-24 18:09:45,423 - __main__ - INFO - Exiting! Moonraker connection lost!
2022-02-24 18:10:05,384 - __main__ - ERROR - Handshake status 401 Unauthorized
2022-02-24 18:10:05,385 - __main__ - INFO - WebSocket closed
2022-02-24 18:10:05,385 - __main__ - INFO - Exiting! Moonraker connection lost!
2022-02-24 18:10:25,337 - __main__ - ERROR - Handshake status 401 Unauthorized
2022-02-24 18:10:25,338 - __main__ - INFO - WebSocket closed
2022-02-24 18:10:25,339 - __main__ - INFO - Exiting! Moonraker connection lost!
2022-02-24 18:10:45,072 - __main__ - ERROR - Handshake status 401 Unauthorized
2022-02-24 18:10:45,073 - __main__ - INFO - WebSocket closed
2022-02-24 18:10:45,073 - __main__ - INFO - Exiting! Moonraker connection lost!
2022-02-24 18:11:04,935 - __main__ - ERROR - Handshake status 401 Unauthorized
2022-02-24 18:11:04,935 - __main__ - INFO - WebSocket closed
2022-02-24 18:11:04,935 - __main__ - INFO - Exiting! Moonraker connection lost!

### If needed, paste the relevant bot journal contents here.

-- Logs begin at Wed 2022-02-23 19:45:43 CST, end at Thu 2022-02-24 18:10:26 CST. --
Feb 24 18:10:25 fluiddpi python[28640]: 2022-02-24 18:10:25,339 - apscheduler.scheduler - INFO - Scheduler has been shut down
[... 25 lines cut ...]
Feb 24 18:10:03 fluiddpi python[28513]: 2022-02-24 18:10:03,542 - apscheduler.scheduler - INFO - Scheduler started
Feb 24 18:10:00 fluiddpi systemd[1]: Started Starts Moonraker Telegram Bot on startup.
Feb 24 18:10:00 fluiddpi systemd[1]: Stopped Starts Moonraker Telegram Bot on startup.
Feb 24 18:10:00 fluiddpi systemd[1]: moonraker-telegram-bot.service: Scheduled restart job, restart counter is at 42.
Feb 24 18:10:00 fluiddpi systemd[1]: moonraker-telegram-bot.service: Service RestartSec=5s expired, scheduling restart.
Feb 24 18:09:55 fluiddpi systemd[1]: moonraker-telegram-bot.service: Succeeded.
Feb 24 18:09:53 fluiddpi python[28387]: 2022-02-24 18:09:53,416 - apscheduler.executors.default - INFO - Job "reshedule (trigger: interval[0:00:02], next run at: 2022-02-24
Feb 24 18:09:53 fluiddpi python[28387]: 2022-02-24 18:09:53,416 - apscheduler.executors.default - INFO - Running job "reshedule (trigger: interval[0:00:02], next run at: 20
Feb 24 18:09:51 fluiddpi python[28387]: 2022-02-24 18:09:51,417 - apscheduler.executors.default - INFO - Job "reshedule (trigger: interval[0:00:02], next run at: 2022-02-24
Feb 24 18:09:51 fluiddpi python[28387]: 2022-02-24 18:09:51,415 - apscheduler.executors.default - INFO - Running job "reshedule (trigger: interval[0:00:02], next run at: 20
Feb 24 18:09:49 fluiddpi python[28387]: 2022-02-24 18:09:49,420 - apscheduler.executors.default - INFO - Job "reshedule (trigger: interval[0:00:02], next run at: 2022-02-24
Feb 24 18:09:49 fluiddpi python[28387]: 2022-02-24 18:09:49,418 - apscheduler.executors.default - INFO - Running job "reshedule (trigger: interval[0:00:02], next run at: 20
Feb 24 18:09:47 fluiddpi python[28387]: 2022-02-24 18:09:47,415 - apscheduler.executors.default - INFO - Job "reshedule (trigger: interval[0:00:02], next run at: 2022-02-24
Feb 24 18:09:47 fluiddpi python[28387]: 2022-02-24 18:09:47,414 - apscheduler.executors.default - INFO - Running job "reshedule (trigger: interval[0:00:02], next run at: 20
Feb 24 18:09:45 fluiddpi python[28387]: 2022-02-24 18:09:45,423 - apscheduler.scheduler - INFO - Scheduler has been shut down
Feb 24 18:09:45 fluiddpi python[28387]: 2022-02-24 18:09:45,423 - __main__ - INFO - Exiting! Moonraker connection lost!
Feb 24 18:09:45 fluiddpi python[28387]: 2022-02-24 18:09:45,423 - __main__ - INFO - WebSocket closed

### What branch does this occur on?

master

### Reports only on latest master or development branches!

- [X] I have updated to latest development/master version before submitting the bug

teslabol77, 2022-02-27:
> Well, the bot gets an "unauthorized" error when trying to access moonraker. Any ideas why that might be happening?

No idea why. It's a vague error. Any ideas on your end?

nlef, 2022-02-27:
> No idea why. It's a vague error. Any ideas on your end?

You have enabled authorization in moonraker and the moonraker login password is not specified or incorrectly specified in the bot configuration.

nlef, 2022-03-19:
Close issue due to inactivity.
Don't hesitate to reopen if the issue persists.

closed 2022-03-19 by nlef

alber70g, 2024-08-22:
If you have authorization enabled, you can add 127.0.0.1 as trusted host

```
# moonraker.conf

[authorization]

trusted_clients:
  192.168.1.0/24
  127.0.0.1 # for telegram bot

```