nlef/moonraker-telegram-bot #120: [Bug]: Not fully functioning after update to 1.4.0 (reinstall only works for about a week)

radaiko, 2022-03-04:
### What happened and in what context?

Since the update on 1.4.0 or higher the bot has lost many functions.

1. I can request a status with /status but there is will be no temperatures and progress and co is always 0.
3. /cancel is not process. I get the message if I really want it but there will be no execution on the Klipper side.
5. If any error appears I get no notification
6. I have a process_notification in my cfg but no notification will be send.

I remove the bot completely via KIAUH and installed it now. Then it worked for about a week. Today directly after a print finished I got the message "Printer Online" (I didn't request a restart and the PI also didn't restart) and now I have the same problems as mention above.

Only solution is completely delete and reinstall.

### If needed, paste the relevant bot log contents here.

2022-02-27 11:15:47,212 - __main__ - ERROR - Uncaught exception
Traceback (most recent call last):
  File "/home/pi/moonraker-telegram-bot/bot/main.py", line 919, in <module>
    greeting_message()
  File "/home/pi/moonraker-telegram-bot/bot/main.py", line 512, in greeting_message
    bot_updater.bot.set_my_commands(commands=commands)
  File "/home/pi/moonraker-telegram-bot-env/lib/python3.7/site-packages/telegram/bot.py", line 130, in decorator
    result = func(*args, **kwargs)
  File "/home/pi/moonraker-telegram-bot-env/lib/python3.7/site-packages/telegram/bot.py", line 5488, in set_my_commands
    result = self._post('setMyCommands', data, timeout=timeout, api_kwargs=api_kwargs)
  File "/home/pi/moonraker-telegram-bot-env/lib/python3.7/site-packages/telegram/bot.py", line 296, in _post
    f'{self.base_url}/{endpoint}', data=data, timeout=effective_timeout
  File "/home/pi/moonraker-telegram-bot-env/lib/python3.7/site-packages/telegram/utils/request.py", line 366, in post
    **urlopen_kwargs,
  File "/home/pi/moonraker-telegram-bot-env/lib/python3.7/site-packages/telegram/utils/request.py", line 279, in _request_wrapper
    raise BadRequest(message)
telegram.error.BadRequest: Command length must not exceed 32
Stack (most recent call last):
  File "/home/pi/moonraker-telegram-bot/bot/main.py", line 55, in handle_exception
    logger.error("Uncaught exception", exc_info=(exc_type, exc_value, exc_traceback), stack_info=True)

### If needed, paste the relevant bot journal contents here.

Mar 04 20:09:31 BedSlinger python[609]:     logger.error("Uncaught exception", exc_info=(exc_type, exc_value, exc_traceback), stack_info=True)
Mar 04 20:09:31 BedSlinger python[609]:   File "/home/pi/moonraker-telegram-bot/bot/main.py", line 55, in handle_exception
Mar 04 20:09:31 BedSlinger python[609]: Stack (most recent call last):
Mar 04 20:09:31 BedSlinger python[609]: telegram.error.BadRequest: Command length must not exceed 32
Mar 04 20:09:31 BedSlinger python[609]:     raise BadRequest(message)
Mar 04 20:09:31 BedSlinger python[609]:   File "/home/pi/moonraker-telegram-bot-env/lib/python3.7/site-packages/telegram/utils/request.py", line 279, in _request_wrapper
Mar 04 20:09:31 BedSlinger python[609]:     **urlopen_kwargs,
Mar 04 20:09:31 BedSlinger python[609]:   File "/home/pi/moonraker-telegram-bot-env/lib/python3.7/site-packages/telegram/utils/request.py", line 366, in post
Mar 04 20:09:31 BedSlinger python[609]:     f'{self.base_url}/{endpoint}', data=data, timeout=effective_timeout
Mar 04 20:09:31 BedSlinger python[609]:   File "/home/pi/moonraker-telegram-bot-env/lib/python3.7/site-packages/telegram/bot.py", line 296, in _post
Mar 04 20:09:31 BedSlinger python[609]:     result = self._post('setMyCommands', data, timeout=timeout, api_kwargs=api_kwargs)
Mar 04 20:09:31 BedSlinger python[609]:   File "/home/pi/moonraker-telegram-bot-env/lib/python3.7/site-packages/telegram/bot.py", line 5488, in set_my_commands
Mar 04 20:09:31 BedSlinger python[609]:     result = func(*args, **kwargs)
Mar 04 20:09:31 BedSlinger python[609]:   File "/home/pi/moonraker-telegram-bot-env/lib/python3.7/site-packages/telegram/bot.py", line 130, in decorator
Mar 04 20:09:31 BedSlinger python[609]:     bot_updater.bot.set_my_commands(commands=commands)
Mar 04 20:09:31 BedSlinger python[609]:   File "/home/pi/moonraker-telegram-bot/bot/main.py", line 512, in greeting_message
Mar 04 20:09:31 BedSlinger python[609]:     greeting_message()
Mar 04 20:09:31 BedSlinger python[609]:   File "/home/pi/moonraker-telegram-bot/bot/main.py", line 919, in <module>
Mar 04 20:09:31 BedSlinger python[609]: Traceback (most recent call last):
Mar 04 20:09:31 BedSlinger python[609]: 2022-03-04 20:09:31,201 - __main__ - ERROR - Uncaught exception
Mar 04 20:09:30 BedSlinger python[609]: 2022-03-04 20:09:30,835 - apscheduler.scheduler - INFO - Scheduler started
Mar 04 20:09:30 BedSlinger python[609]: 2022-03-04 20:09:30,635 - apscheduler.scheduler - INFO - Scheduler started
Mar 04 20:09:20 BedSlinger systemd[1]: Started Starts Moonraker Telegram Bot on startup.

### What branch does this occur on?

master

### Reports only on latest master or development branches!

- [X] I have updated to latest development/master version before submitting the bug

radaiko, 2022-03-04:
This is what it looks before it broke:
[image]

This is after:
[image]

The second pictures was forced with /status. The first one was automatically updated.

radaiko, 2022-03-05:
@nlef: found the problem. Only 32 commands are allowed but I got to many custom macros (because of custom MMU).

So to fix this problem I had to add "include_macros_in_command_list: false" to my config. 
In the bot/main.py the limit is set to 100, so maybe this should be changed to 32.

nlef, 2022-03-05:
> @nlef: found the problem. Only 32 commands are allowed but I got to many custom macros (because of custom MMU).
> 
> So to fix this problem I had to add "include_macros_in_command_list: false" to my config. In the bot/main.py the limit is set to 100, so maybe this should be changed to 32.
@radaiko 
could you provide a link to the documentation where the limit is 32 commands? I found a limit of 100 commands.
please provide a list of macro names, perhaps they contain symbols that lead to errors

radaiko, 2022-03-05:
> @nlef: found the problem. Only 32 commands are allowed but I got to many custom macros (because of custom MMU).
>
> So to fix this problem I had to add "include_macros_in_command_list: false" to my [...]

@nlef : I read it wrong. It's not 32 commands in the length of the command. So only 32 char per name.
I got a macro which was called "ERCF_UNLOAD_FILAMENT_IN_EXTRUDER_WITH_TIP_FORMING" after cutting it to 32 characters it was working. The reason why it worked sometimes was I was starting the telegram bot without that macro in the cfg and added it later to the printer.cfg. When only restarting the firmware without a telegram bot it was working.

Maybe a check for the length of a macro would be good.

nlef, 2022-03-08:
@radaiko please try the new version.

radaiko, 2022-03-09:
@nlef tested with 1.4.2 still the same problem

`2022-03-09 20:21:14,976 - __main__ - ERROR - Uncaught exception
Traceback (most recent call last):
  File "/home/pi/moonraker-telegram-bot/bot/main.py", line 1349, in <module>
    greeting_message()
  File "/home/pi/moonraker-telegram-bot/bot/main.py", line 906, in greeting_message
    bot_updater.bot.set_my_commands(commands=commands)
  File "/home/pi/moonraker-telegram-bot-env/lib/python3.7/site-packages/telegram/bot.py", line 130, in decorator
    result = func(*args, **kwargs)
  File "/home/pi/moonraker-telegram-bot-env/lib/python3.7/site-packages/telegram/bot.py", line 5488, in set_my_commands
    result = self._post('setMyCommands', data, timeout=timeout, api_kwargs=api_kwargs)
  File "/home/pi/moonraker-telegram-bot-env/lib/python3.7/site-packages/telegram/bot.py", line 296, in _post
    f'{self.base_url}/{endpoint}', data=data, timeout=effective_timeout
  File "/home/pi/moonraker-telegram-bot-env/lib/python3.7/site-packages/telegram/utils/request.py", line 366, in post
    **urlopen_kwargs,
  File "/home/pi/moonraker-telegram-bot-env/lib/python3.7/site-packages/telegram/utils/request.py", line 279, in _request_wrapper
    raise BadRequest(message)
telegram.error.BadRequest: Command length must not exceed 32
Stack (most recent call last):
  File "/home/pi/moonraker-telegram-bot/bot/main.py", line 54, in handle_exception
    stack_info=True,`

nlef, 2022-03-09:
@nlef it's my fault, didn't test well. checked and pushed a small update

nlef, 2022-03-19:
@radaiko 
is the error still present?

radaiko, 2022-03-26:
@nlef

Sorry for the late response. Was a little busy with printing V2.4 parts.

With 1.4.2-1 this is fixed.

closed 2022-03-26 by radaiko