nlef/moonraker-telegram-bot #121: Simple Mode aka customization of notifications

Technofrikus, 2022-03-06:
### What would you like to see added?

I would like to have an option where I only get two types of messages:
- Errors
- Print is finished

I was not able to configure it this way with the current options. I know when I am starting a print, so I dont need a notification for that. Or am I misunderstanding the settings?

b00gie-dev, 2022-03-08:
That's exactly my problem too. Before the updates to the single notification this has worked fine. But now I am also not able to get notifications when the print is done.

aka13-404, 2022-04-09:
> That's exactly my problem too. Before the updates to the single notification this has worked fine. But now I am also not able to get notifications when the print is done.

If you want an additional message that the print is completed, I suggest perhaps to add a [notification](https://github.com/nlef/moonraker-telegram-bot/wiki/Interacting-with-klipper#send-custom-notifications-to-the-bot) to your end g-code, would that solve your issue?

> ### What would you like to see added?
>
> I would like to have an option where I only get two types of messages:
>
>     * Errors
>
>     * Print is finished
>
>
> I was not able to configure it this way with [...]

We have doublechecked our internal logic, you are right, it would be logical to not receive any progress messages with the disabled [progress_notification] section. I'll refer to this ticket when this is fixed, then you can as well add a single line to your endcode, to get a notification on print end.

Sorry it took so long, we had more urgent issues to fix.

nlef, 2022-04-09:
fixed in 98d5fe9bf1577d8951df8691f8889265e26c79c5

Technofrikus, 2022-04-09:
Just adding a gcode message is a good idea! Thanks.

And thanks for fixing this and the bot itself! Great help!

closed 2022-05-05 by aka13-404

Technofrikus, 2022-06-02:
I cant seem do disable notifications. I always get the start message, start timelapse and print finished message. I do want the finish messeage, but not the other ones. The timelapse never worked for me either, but thats another problem and not important for me.

```
[bot]
server: localhost
chat_id: xxx
bot_token: xxx
log_path: /home/pi/klipper_logs

[progress_notification]
# percent: 100

[telegram_ui]
silent_status: true
silent_progress: true

[camera]
host: http://localhost:8080/?action=stream
fps:10

[timelapse]
send_finished_lapse: true
basedir: /home/pi/klipper_logs/timelapse

```