nlef/moonraker-telegram-bot #122: [BUG] The bot is SUPER agressive when it cant hit moonraker

simpat1zq, 2022-03-10:
### What happened and in what context?

So I was doing updates and I updated telegram to the latest version and that went fine. Then I updated moonraker and it had some issues I had to work through. And while it was down, telegram sent me a message letting me know. It seemed like the messages were at least every 30 seconds, maybe a little faster. It might be a good idea to just limit it to one message when it doesn't connect. Or at least once every 30 minutes or so. 

### What branch does this occur on?

master

### Reports only on latest master or development branches!

- [X] I have updated to latest development/master version before submitting the bug

aka13-404, 2022-05-05:
This is something which definately will need some attention in the next release

aka13-404, 2024-11-25:
Fixed in 2.0.0

closed 2024-11-25 by aka13-404