nlef/moonraker-telegram-bot #123: Log filling with errors

Prutsium, 2022-03-13:
Version: Latest git
Since a while it doesnt work anymore for me. I had it working very well but now my log got filled with tons of errors.
2022-03-13 11:20:28,258 - __main__ - ERROR - Uncaught exception
Traceback (most recent call last):
  File "/home/pi/moonraker-telegram-bot/bot/main.py", line 1349, in <module>
    greeting_message()
  File "/home/pi/moonraker-telegram-bot/bot/main.py", line 906, in greeting_message
    bot_updater.bot.set_my_commands(commands=commands)
  File "/home/pi/moonraker-telegram-bot-env/lib/python3.7/site-packages/telegram/bot.py", line 130, in decorator
    result = func(*args, **kwargs)
  File "/home/pi/moonraker-telegram-bot-env/lib/python3.7/site-packages/telegram/bot.py", line 5488, in set_my_commands
    result = self._post('setMyCommands', data, timeout=timeout, api_kwargs=api_kwargs)
  File "/home/pi/moonraker-telegram-bot-env/lib/python3.7/site-packages/telegram/bot.py", line 296, in _post
    f'{self.base_url}/{endpoint}', data=data, timeout=effective_timeout
  File "/home/pi/moonraker-telegram-bot-env/lib/python3.7/site-packages/telegram/utils/request.py", line 366, in post
    **urlopen_kwargs,
[telegram.txt](https://github.com/nlef/moonraker-telegram-bot/files/8239404/telegram.txt)

  File "/home/pi/moonraker-telegram-bot-env/lib/python3.7/site-packages/telegram/utils/request.py", line 279, in _request_wrapper
    raise BadRequest(message)
telegram.error.BadRequest: Bot_command_invalid
Stack (most recent call last):
  File "/home/pi/moonraker-telegram-bot/bot/main.py", line 54, in handle_exception
    stack_info=True,

Added the config

nlef, 2022-03-13:
@Prutsium 
judging by the log, the problem is in adding macros to the list of bot commands. You can disable this by setting `include_macros_in_command_list:false` 
Сould you provide a list of macros, maybe there are some characters in their names that telegram does not accept

Prutsium, 2022-03-13:
It would use the default macros or macros specific specified in the config?
These are my macro's in mainsail but did not specific to the bot.
[image]

Prutsium, 2022-03-13:
Ps ... where (under what [] header does the include_macros_in_command_list:false go?

nlef, 2022-03-13:
> Ps ... where (under what [] header does the include_macros_in_command_list:false go?
config section [telegram_ui]
https://github.com/nlef/moonraker-telegram-bot/wiki/Sample-config#telegram_ui

Prutsium, 2022-03-13:
Ok seems to work now but as you see there are no strange macro's or characters in a macro.

nlef, 2022-03-13:
> Ok seems to work now but as you see there are no strange macro's or characters in a macro.

bot updated to latest version on commit 0acc60ebf8bffb5f80f253af69cf67d3d4df65d4 ?

Prutsium, 2022-03-13:
[image]

Prutsium, 2022-03-13:
So yeah it was up to date

nlef, 2022-03-13:
> So yeah it was up to date

Could you please give me the clipper config with macros so that I don't have to rewrite macro names from the screenshot for tests?

Prutsium, 2022-03-13:
Thats the difficult part as it contains many include files :)
[klippy_consilidated.txt](https://github.com/nlef/moonraker-telegram-bot/files/8239613/klippy_consilidated.txt)

Better via the log there they are consolidated
.

nlef, 2022-03-13:
> Thats the difficult part as it contains many include files :) [klippy_consilidated.txt](https://github.com/nlef/moonraker-telegram-bot/files/8239613/klippy_consilidated.txt)
> 
> Better via the log there they are consolidated .

macro list without gcode_shell_command works correctly. testing of the additional module is not planned yet

Prutsium, 2022-03-13:
Anyway i dont need the macro's so for me ok without any macro as it's for pure monitoring only.
But for example Ratrig (RatOs) has per default shell commands in them and it worked before perfectly so seems due some recent change(s) it doesnt work anymore.

nlef, 2022-03-19:
@Prutsium 
I installed "G-Code Shell Command Extension" and tested your macro list. I can't reproduce the error.

closed 2022-05-05 by aka13-404