nlef/moonraker-telegram-bot #124: Not building timelapse

Prutsium, 2022-03-18:
After the Gcode macro's issues in another report i notice an other issue (probably just on my side)

But timelapse is nit getting build anymore. the tmp dir is empty but also the final dir stay's empty.

Log says:
`2022-03-18 17:23:30,032 - __main__ - DEBUG - {"jsonrpc": "2.0", "method": "notify_gcode_response", "params": ["Done printing file"]}
2022-03-18 17:23:30,055 - __main__ - DEBUG - {"jsonrpc": "2.0", "method": "notify_status_update", "params": [{"print_stats": {"state": "complete", "print_duration": 1636.825756134931, "total_duration": 2055.1312416079454}, "display_status": {"progress": 1.0, "message": "Done."}, "gcode_move": {"gcode_position": [150.0, 293.0, 8.040000000000001, 363.5127400000101]}, "virtual_sdcard": {"progress": 1.0}, "extruder": {"temperature": 260.04}, "heater_bed": {"temperature": 110.33}, "temperature_sensor Stepsticks": {"temperature": 26.97}, "temperature_sensor Spider": {"temperature": 36.45}, "temperature_sensor raspberry_pi": {"temperature": 39.92}}, 905599.443947368]}
2022-03-18 17:23:30,059 - apscheduler.scheduler - INFO - Added job "Notifier._notify" to job store "default"
2022-03-18 17:23:30,061 - apscheduler.scheduler - INFO - Added job "Timelapse._send_lapse" to job store "default"
2022-03-18 17:23:30,062 - apscheduler.scheduler - DEBUG - Looking for jobs to run
2022-03-18 17:23:30,064 - apscheduler.scheduler - INFO - Removed job 3fd70fc76173468094af57e35141a0a5
2022-03-18 17:23:30,066 - apscheduler.scheduler - INFO - Removed job 40bd40ddc7e445e78d8d7aa5c588ee1d
2022-03-18 17:23:30,069 - apscheduler.scheduler - DEBUG - No jobs; waiting until a job is added
2022-03-18 17:23:30,068 - apscheduler.executors.default - INFO - Running job "Timelapse._send_lapse (trigger: date[2022-03-18 17:23:30 GMT], next run at: 2022-03-18 17:23:30 GMT)" (scheduled at 2022-03-18 17:23:30.060118+00:00)
2022-03-18 17:23:30,068 - apscheduler.executors.default - INFO - Running job "Notifier._notify (trigger: date[2022-03-18 17:23:30 GMT], next run at: 2022-03-18 17:23:30 GMT)" (scheduled at 2022-03-18 17:23:30.057932+00:00)
2022-03-18 17:23:30,072 - apscheduler.scheduler - DEBUG - Looking for jobs to run
2022-03-18 17:23:30,070 - apscheduler.scheduler - INFO - Added job "Notifier._send_print_finish" to job store "default"
2022-03-18 17:23:30,080 - apscheduler.scheduler - INFO - Removed job b7036ff459664c5c89fce285920495fd
2022-03-18 17:23:30,081 - apscheduler.executors.default - INFO - Running job "Notifier._send_print_finish (trigger: date[2022-03-18 17:23:30 GMT], next run at: 2022-03-18 17:23:30 GMT)" (scheduled at 2022-03-18 17:23:30.063883+00:00)
2022-03-18 17:23:30,082 - apscheduler.scheduler - DEBUG - No jobs; waiting until a job is added
2022-03-18 17:23:30,090 - apscheduler.scheduler - DEBUG - Looking for jobs to run
2022-03-18 17:23:30,091 - apscheduler.scheduler - DEBUG - No jobs; waiting until a job is added
2022-03-18 17:23:30,086 - __main__ - DEBUG - {"jsonrpc": "2.0", "method": "notify_history_changed", "params": [{"action": "finished", "job": {"end_time": 1647624210.0599637, "filament_used": 4049.1235500002695, "filename": "Fabulous Lappi-Amberis(2)_0.2mm_32m.gcode", "metadata": {"size": 2147921, "modified": 1647622154.4174902, "slicer": "SuperSlicer", "slicer_version": "2.3.57", "gcode_start_byte": 55683, "gcode_end_byte": 2133886, "layer_count": 30, "object_height": 6.04, "estimated_time": 1936, "nozzle_diameter": 0.4, "layer_height": 0.2, "first_layer_height": 0.24, "first_layer_extr_temp": 260.0, "first_layer_bed_temp": 110.0, "filament_name": "\"PM ABS-T - RatRig\"", "filament_type": "ABS", "filament_total": 4025.33, "filament_weight_total": 10.46, "thumbnails": [{"width": 32, "height": 32, "size": 1630, "relative_path": ".thumbs/Fabulous Lappi-Amberis(2)_0.2mm_32m-32x32.png"}, {"width": 400, "height": 300, "size": 37933, "relative_path": ".thumbs/Fabulous Lappi-Amberis(2)_0.2mm_32m-400x300.png"}]}, "print_duration": 1636.825756134931, "status": "completed", "start_time": 1647622155.1305168, "total_duration": 2055.1312416079454, "job_id": "000086", "exists": true}}]}
2022-03-18 17:23:30,300 - __main__ - DEBUG - {"jsonrpc": "2.0", "method": "notify_status_update", "params": [{"extruder": {"temperature": 260.0}, "temperature_sensor Stepsticks": {"temperature": 27.14}}, 905599.694315202]}
2022-03-18 17:23:30,555 - __main__ - DEBUG - {"jsonrpc": "2.0", "method": "notify_status_update", "params": [{"extruder": {"temperature": 259.91}, "heater_bed": {"temperature": 110.34}, "temperature_sensor Stepsticks": {"temperature": 27.11}, "temperature_sensor Spider": {"temperature": 36.65}}, 905599.944821146]}
2022-03-18 17:23:30,782 - __main__ - DEBUG - {"jsonrpc": "2.0", "method": "notify_proc_stat_update", "params": [{"moonraker_stats": {"time": 1647624210.7807205, "cpu_usage": 7.29, "memory": 32660, "mem_units": "kB"}, "cpu_temp": 40.407, "network": {"lo": {"rx_bytes": 52565548145, "tx_bytes": 52565548145, "bandwidth": 1217793.95}, "eth0": {"rx_bytes": 1978844075, "tx_bytes": 3473726920, "bandwidth": 16532.83}, "wlan0": {"rx_bytes": 0, "tx_bytes": 0, "bandwidth": 0.0}}, "system_cpu_usage": {"cpu": 33.84, "cpu0": 25.26, "cpu1": 10.2, "cpu2": 41.24, "cpu3": 57.28}, "websocket_connections": 3}]}
2022-03-18 17:23:30,802 - __main__ - DEBUG - {"jsonrpc": "2.0", "method": "notify_status_update", "params": [{"heater_bed": {"temperature": 110.33}, "temperature_sensor Stepsticks": {"temperature": 27.17}, "temperature_sensor Spider": {"temperature": 36.49}}, 905600.195626794]}
2022-03-18 17:23:31,052 - __main__ - DEBUG - {"jsonrpc": "2.0", "method": "notify_status_update", "params": [{"extruder": {"temperature": 259.69}, "heater_bed": {"temperature": 110.32}, "temperature_sensor Stepsticks": {"temperature": 27.14}, "temperature_sensor Spider": {"temperature": 36.57}, "temperature_sensor raspberry_pi": {"temperature": 41.38}}, 905600.446600034]}
2022-03-18 17:23:31,302 - __main__ - DEBUG - {"jsonrpc": "2.0", "method": "notify_status_update", "params": [{"extruder": {"temperature": 259.64}, "heater_bed": {"temperature": 110.29}, "temperature_sensor Spider": {"temperature": 36.61}}, 905600.697554108]}
2022-03-18 17:23:31,554 - __main__ - DEBUG - {"jsonrpc": "2.0", "method": "notify_status_update", "params": [{"extruder": {"temperature": 259.48}, "heater_bed": {"temperature": 110.25}, "temperature_sensor Spider": {"temperature": 36.77}}, 905600.948586164]}
2022-03-18 17:23:31,785 - __main__ - DEBUG - {"jsonrpc": "2.0", "method": "notify_proc_stat_update", "params": [{"moonraker_stats": {"time": 1647624211.783074, "cpu_usage": 2.79, "memory": 32660, "mem_units": "kB"}, "cpu_temp": 42.355, "network": {"lo": {"rx_bytes": 52566057452, "tx_bytes": 52566057452, "bandwidth": 1017305.77}, "eth0": {"rx_bytes": 1978845262, "tx_bytes": 3473733630, "bandwidth": 7886.86}, "wlan0": {"rx_bytes": 0, "tx_bytes": 0, "bandwidth": 0.0}}, "system_cpu_usage": {"cpu": 53.94, "cpu0": 32.65, "cpu1": 98.0, "cpu2": 14.81, "cpu3": 74.75}, "websocket_connections": 3}]}
2022-03-18 17:23:31,806 - __main__ - DEBUG - {"jsonrpc": "2.0", "method": "notify_status_update", "params": [{"extruder": {"temperature": 259.3}, "temperature_sensor Stepsticks": {"temperature": 26.86}}, 905601.199287553]}
2022-03-18 17:23:32,055 - __main__ - DEBUG - {"jsonrpc": "2.0", "method": "notify_status_update", "params": [{"extruder": {"temperature": 259.03}, "heater_bed": {"temperature": 110.21}, "temperature_sensor Stepsticks": {"temperature": 27.04}, "temperature_sensor Spider": {"temperature": 36.61}, "temperature_sensor raspberry_pi": {"temperature": 41.87}}, 905601.449652441]}
2022-03-18 17:23:32,306 - __main__ - DEBUG - {"jsonrpc": "2.0", "method": "notify_status_update", "params": [{"extruder": {"temperature": 258.78}, "heater_bed": {"temperature": 110.15}, "temperature_sensor Stepsticks": {"temperature": 27.07}, "temperature_sensor Spider": {"temperature": 36.49}}, 905601.700670108]}
2022-03-18 17:23:32,556 - __main__ - DEBUG - {"jsonrpc": "2.0", "method": "notify_status_update", "params": [{"heater_bed": {"temperature": 110.09}, "temperature_sensor Stepsticks": {"temperature": 27.04}, "temperature_sensor Spider": {"temperature": 36.61}}, 905601.951268367]}
2022-03-18 17:23:32,784 - __main__ - DEBUG - {"jsonrpc": "2.0", "method": "notify_proc_stat_update", "params": [{"moonraker_stats": {"time": 1647624212.78297, "cpu_usage": 2.67, "memory": 32660, "mem_units": "kB"}, "cpu_temp": 40.894, "network": {"lo": {"rx_bytes": 52566075480, "tx_bytes": 52566075480, "bandwidth": 36020.06 [... cut ...]
  File "/home/pi/moonraker-telegram-bot-env/lib/python3.7/site-packages/telegram/bot.py", line 296, in _post
    f'{self.base_url}/{endpoint}', data=data, timeout=effective_timeout
  File "/home/pi/moonraker-telegram-bot-env/lib/python3.7/site-packages/telegram/utils/request.py", line 359, in post
    result = self._request_wrapper('POST', url, fields=data, **urlopen_kwargs)
  File "/home/pi/moonraker-telegram-bot-env/lib/python3.7/site-packages/telegram/utils/request.py", line 279, in _request_wrapper
    raise BadRequest(message)
telegram.error.BadRequest: Canceled by new editmessagemedia request`

nlef, 2022-03-28:
@Prutsium 
it looks like the directory `/home/pi/timelapse-tmp/Fabulous Lappi-Amberis(2)_0.2mm_32m.gcode_2022-03-18_16-49/` was not created. does this directory `/home/pi/timelapse-tmp/` exist? whether the user who runs the bot has write permissions?

Prutsium, 2022-03-29:
Hi,
Have checked this and seems all runs under the user pi
The bot runs as pi:

And the directory's

So doesnt seem an rights issue.

#124)

aka13-404, 2022-05-05:
Is this still happening?

Prutsium, 2022-05-05:
Ah i gave AK feedback that with the Dev version all works ok now.

#124)

closed 2022-05-05 by aka13-404