nlef/moonraker-telegram-bot #126: Not getting Status Updates

niketannir, 2022-03-26:
### What happened and in what context?

Not getting Status Updates as per the set value, and not getting updates when requested. When I run the /status command in telegram I dont get any reply. When a print is in progress I do not get the status updates every 5% as set in the config.

Please help as Im not sure what the issue could be.

### If needed, paste the relevant bot log contents here.

2022-03-26 19:29:50,892 - __main__ - ERROR - Exception while handling an update:
Traceback (most recent call last):
  File "/home/pi/moonraker-telegram-bot-env/lib/python3.7/site-packages/telegram/ext/utils/promise.py", line 96, in run
    self._result = self.pooled_function(*self.args, **self.kwargs)
  File "/home/pi/moonraker-telegram-bot/bot/main.py", line 141, in status
    mess = escape_markdown(klippy.get_status(), version=2)
  File "/home/pi/moonraker-telegram-bot/bot/klippy.py", line 383, in get_status
    message += self._get_printing_file_info()
  File "/home/pi/moonraker-telegram-bot/bot/klippy.py", line 336, in _get_printing_file_info
    eta = self._get_eta()
  File "/home/pi/moonraker-telegram-bot/bot/klippy.py", line 292, in _get_eta
    eta = int(self.printing_duration / self.vsd_progress - self.printing_duration)
ZeroDivisionError: float division by zero
2022-03-26 19:34:08,609 - __main__ - ERROR - Exception while handling an update:
Traceback (most recent call last):
  File "/home/pi/moonraker-telegram-bot-env/lib/python3.7/site-packages/telegram/ext/utils/promise.py", line 96, in run
    self._result = self.pooled_function(*self.args, **self.kwargs)
  File "/home/pi/moonraker-telegram-bot/bot/main.py", line 141, in status
    mess = escape_markdown(klippy.get_status(), version=2)
  File "/home/pi/moonraker-telegram-bot/bot/klippy.py", line 383, in get_status
    message += self._get_printing_file_info()
  File "/home/pi/moonraker-telegram-bot/bot/klippy.py", line 336, in _get_printing_file_info
    eta = self._get_eta()
  File "/home/pi/moonraker-telegram-bot/bot/klippy.py", line 292, in _get_eta
    eta = int(self.printing_duration / self.vsd_progress - self.printing_duration)
ZeroDivisionError: float division by zero

### If needed, paste the relevant bot journal contents here.

-- Logs begin at Fri 2022-03-25 21:17:01 GMT, end at Sat 2022-03-26 19:43:42 GMT
Mar 26 19:34:08 raspberrypi python[16867]: ZeroDivisionError: float division by
Mar 26 19:34:08 raspberrypi python[16867]:     eta = int(self.printing_duration
Mar 26 19:34:08 raspberrypi python[16867]:   File "/home/pi/moonraker-telegram-b
Mar 26 19:34:08 raspberrypi python[16867]:     eta = self._get_eta()
Mar 26 19:34:08 raspberrypi python[16867]:   File "/home/pi/moonraker-telegram-b
Mar 26 19:34:08 raspberrypi python[16867]:     message += self._get_printing_fil
Mar 26 19:34:08 raspberrypi python[16867]:   File "/home/pi/moonraker-telegram-b
Mar 26 19:34:08 raspberrypi python[16867]:     mess = escape_markdown(klippy.get
Mar 26 19:34:08 raspberrypi python[16867]:   File "/home/pi/moonraker-telegram-b
Mar 26 19:34:08 raspberrypi python[16867]:     self._result = self.pooled_functi
Mar 26 19:34:08 raspberrypi python[16867]:   File "/home/pi/moonraker-telegram-b
Mar 26 19:34:08 raspberrypi python[16867]: Traceback (most recent call last):
Mar 26 19:34:08 raspberrypi python[16867]: 2022-03-26 19:34:08,609 - __main__ -
Mar 26 19:29:50 raspberrypi python[16867]: ZeroDivisionError: float division by
Mar 26 19:29:50 raspberrypi python[16867]:     eta = int(self.printing_duration
Mar 26 19:29:50 raspberrypi python[16867]:   File "/home/pi/moonraker-telegram-b
Mar 26 19:29:50 raspberrypi python[16867]:     eta = self._get_eta()
Mar 26 19:29:50 raspberrypi python[16867]:   File "/home/pi/moonraker-telegram-b
Mar 26 19:29:50 raspberrypi python[16867]:     message += self._get_printing_fil
Mar 26 19:29:50 raspberrypi python[16867]:   File "/home/pi/moonraker-telegram-b
Mar 26 19:29:50 raspberrypi python[16867]:     mess = escape_markdown(klippy.get
Mar 26 19:29:50 raspberrypi python[16867]:   File "/home/pi/moonraker-telegram-b

### What branch does this occur on?

master

### Reports only on latest master or development branches!

- [X] I have updated to latest development/master version before submitting the bug

nlef, 2022-03-28:
@niketannir 
could you give me your bot config?

niketannir, 2022-03-28:
1 |   | [bot]
-- | -- | --
2 |   | server: 10.0.0.26:7125
3 |   | bot_token: 50#####5:AAGoy###################hlPuE
4 |   | chat_id: 8#####9
5 |   | log_path: /home/pi/klipper_logs
6 |   |  
7 |   | #===================================
8 |   | [camera]
9 |   | host: http://10.0.0.26/webcam/?action=stream
10 |   | fps: 30
11 |   | flip_vertically: false
12 |   | flip_horizontally: false
13 |   | fourcc: x264
14 |   | video_duration: 60
15 |   | video_buffer_size: 3
16 |   | picture_quality: high
17 |   |  
18 |   | #===================================
19 |   |  
20 |   | [progress_notification]
21 |   | percent: 5
22 |   | #time: 900
23 |   |  
24 |   | #===================================
25 |   |  
26 |   | [telegram_ui]
27 |   | status_single_message: true
28 |   | eta_source: slicer
29 |   | buttons: [status,pause,cancel,resume],[files,emergency,macros,shutdown]
30 |   | require_confirmation_macro: true
31 |   | silent_progress: false
32 |   | silent_commands: false
33 |   | silent_status: false
34 |   | include_macros_in_command_list: true
35 |   | show_hidden_macros: false
36 |   | status_message_content: progress, height, filament_length, filament_weight, print_duration, eta, finish_time, m117_status, last_update_time
37 |   | status_message_sensors: mcu_temp, raspberry_pi, chamber_temp
38 |   | status_message_heaters: extruder, heater_bed
39 |   | status_message_temperature_fans: hotend_fan, chamber_fan
40 |   |  
41 |   | #===================================

niketannir, 2022-03-28:
Apologies for the formatting, it was not allowing me to attach the conf file.

niketannir, 2022-03-29:
[telegram 29.03.22.txt](https://github.com/nlef/moonraker-telegram-bot/files/8368953/telegram.29.03.22.txt)

Please see txt file of config

nlef, 2022-04-02:
@niketannir 
some fixes in 284249b07b46e909d61bc52e51d97e073c08e610
pls test)

niketannir, 2022-04-02:
Do I just delete the old files and paste the new ones in?

nlef, 2022-04-02:
> Do I just delete the old files and paste the new ones in?

it's better to switch branch from master to development
Or wait for the next release.

niketannir, 2022-04-03:
Do I just replace https://github.com/nlef/moonraker-telegram-bot.git with https://github.com/nlef/moonraker-telegram-bot/development.git  in my moonraker config?

nlef, 2022-04-03:
> Do I just replace https://github.com/nlef/moonraker-telegram-bot.git with https://github.com/nlef/moonraker-telegram-bot/development.git in my moonraker config?

No. go to bot install dir and enter
`git checkout development`
after that update bot using fluid/mainsail webui)

niketannir, 2022-04-03:
Sorry for being so naive can you tell me where i can locate the bot install directory?

nlef, 2022-04-03:
> Sorry for being so naive can you tell me where i can locate the bot install directory?

~/moonraker-telegram-bot

svcsectool, 2022-04-30:
I can test this out @nlef - I am switching to `development` branch

aka13-404, 2022-05-05:
How is this issue doing, is this still an issue?

closed 2022-06-02 by aka13-404