nlef/moonraker-telegram-bot #128: Multiple bot instances guide

morganfw, 2022-04-02:
### What would you like to see added?

Hello,
I have two instances of Moonraker on the same Raspberry Pi connected to two 3D printers, but I didn’t find any information on the wiki about how to install multiple Moonraker-telegram-bots.

Can you help me understand how to do this and provide a tutorial on how to install multiple instances?

Thank you in advance.

nlef, 2022-04-02:
@morganfw 
you can test the new installer in branch `multiple-instances`

morganfw, 2022-04-02:
Thank you for answer.
I've downloaded multiple-instances branch then tried to install, my klipper_config have 2 paths "ender" and "tronxy", but when I try to select configuration file path, the script try to add "printer_1" that not exists instead of path declared by me (eg: ender):

```
git clone -b multiple-instances https://github.com/nlef/moonraker-telegram-bot.git

cd moonraker-telegram-bot
 
 ./scripts/install.sh
/=======================================================\
 1 Mooonraker instance was found!| Usually you need one Moonraker telegram bot instance per Mooonraker   |
| instance. Though you can install as many as you wish. |
\=======================================================/

###### Number of Moonraker telegram bot instances to set up: 2

###### Install 2 instance(s)? (Y/n): y
###### > Yes
Installing Moonraker telegram bot ...

[...]

Enter the path for the configuration file location.
Its recommended to store it together with the klipper configuration for easier backup and usage.
Enter desired path: /home/pi/klipper_config/ender/

###### Bot configuration file will be located in /home/pi/klipper_config/ender

###### Creating base config file
cp: cannot create regular file '/home/pi/klipper_config/ender/printer_1/telegram.conf': No such file or directory
```

The error is in line 61 here: https://github.com/nlef/moonraker-telegram-bot/blob/multiple-instances/scripts/install.sh#L61 the variable cannot manage path correctly.

May you can fix the script to manage path inserted by user and maybe rename service and log with path name in while loop (eg: "moonraker-telegram-bot-ender.service" and "telegram-logs-ender")?

Thank you in advance.

morganfw, 2022-04-02:
After #130 commit the script works but creates only the first instance and both services, also creates two folders "printer_1" and "printer_2" within the first instance.
No request to configure the second instance.

```
[...]
Enter the path for the configuration file location.
Its recommended to store it together with the klipper configuration for easier backup and usage.
Enter desired path: /home/pi/klipper_config/ender

###### Bot configuration file will be located in /home/pi/klipper_config/ender

###### Creating base config file
[sudo] password for pi:

Created symlink /etc/systemd/system/multi-user.target.wants/moonraker-telegram-bot-1.service → /etc/systemd/system/moonraker-telegram-bot-1.service.

###### moonraker-telegram-bot-1.service instance created!

###### Launching moonraker-telegram-bot instance ...

###### Creating base config file
Created symlink /etc/systemd/system/multi-user.target.wants/moonraker-telegram-bot-2.service → /etc/systemd/system/moonraker-telegram-bot-2.service.

###### moonraker-telegram-bot-2.service instance created!

###### Launching moonraker-telegram-bot instance ...
```

nlef, 2022-04-02:
@morganfw 
there's another commit with custom instance names)

morganfw, 2022-04-02:
I've tried last script but it creates path printer_1 and printer_2 outside custom paths, IMHO for convenience is better to let the user choose the path as in the previous script.

Now I've configured the file "$HOME/klipper_config/printer_1/telegram.conf" then restarted bot, but an error appears even bot_id and chat_id are correct:

```
2022-04-02 23:30:20,190 - __main__ - ERROR - Uncaught exception
Traceback (most recent call last):
  File "/home/pi/moonraker-telegram-bot/bot/main.py", line 1382, in <module>
    bot_updater = start_bot(configWrap.bot.token, configWrap.bot.socks_proxy)
  File "/home/pi/moonraker-telegram-bot/bot/main.py", line 958, in start_bot
    request_kwargs=request_kwargs,
  File "/home/pi/moonraker-telegram-bot-env/lib/python3.7/site-packages/telegram/ext/updater.py", line 290, in __init__
    else arbitrary_callback_data
  File "/home/pi/moonraker-telegram-bot-env/lib/python3.7/site-packages/telegram/ext/extbot.py", line 103, in __init__
    private_key_password=private_key_password,
  File "/home/pi/moonraker-telegram-bot-env/lib/python3.7/site-packages/telegram/bot.py", line 191, in __init__
    self.token = self._validate_token(token)
  File "/home/pi/moonraker-telegram-bot-env/lib/python3.7/site-packages/telegram/bot.py", line 355, in _validate_token
    raise InvalidToken()
telegram.error.InvalidToken: Invalid token
Stack (most recent call last):
  File "/home/pi/moonraker-telegram-bot/bot/main.py", line 54, in handle_exception
    stack_info=True
```,

nlef, 2022-04-02:
> I've tried last script but it creates path printer_1 and printer_2 outside custom paths, IMHO for convenience is better to let the user choose the path as in the previous script.
> 
did you choose `n` for `Use automatic paths? (Y/n):` question? automatic paths will create printer names alike `printer_1` same as kiauh installes moonraker and klipper

morganfw, 2022-04-02:
I don't use KIAUH for klipper and moonraker installations, and before I choose `y` for automatic paths.
After cleaned all and choose `n` for automatic paths, then select my paths and script creates "printer_ender" and "printer_tronxy" path under "$HOME/klipper_config" path that's different from my paths (my paths are "ender" and "tronxy" without "printer_" suffix), but now telegram bot multi instance works fine.

Thank you for support!!!

nlef, 2022-04-03:
@morganfw 
I fixed the creation of folder names (removed the prefix `printer`). Now you can set the base directory where directories for individual instances will be created. and then set custom instance names.

morganfw, 2022-04-03:
@nlef 
ok after clean all and re-execute all the procedure, all works fine!

Now we can insert custom path where common configuration files are located and `telegram.conf` file will be created there, also `klipper_logs` path create `telegram-logs-${custom_instance}` folder with correct custom instance name and service `moonraker-telegram-bot-${custom_instance}.service`

Below my steps:

```
$ git clone -b multiple-instances https://github.com/nlef/moonraker-telegram-bot.git
 
$ cd moonraker-telegram-bot
 
$ ./scripts/install.sh
/=======================================================\
| 1 Mooonraker instance was found!                      |
| Usually you need one Moonraker telegram bot instance per Mooonraker   |
| instance. Though you can install as many as you wish. |
\=======================================================/

###### Number of Moonraker telegram bot instances to set up: 2

###### Install 2 instance(s)? (Y/n): y
###### > Yes

[...]

###### Telegram bot configuration file location selection

Enter the path for the configuration files location. Subfolders for multiple instances wil be created under this path.
Its recommended to store it together with the klipper configuration for easier backup and usage.
Enter desired path: /home/pi/klipper_config

###### Bot configuration file will be located in /home/pi/klipper_config
Use automatic paths? (Y/n): n
###### > No

###### Telegram bot instance name selection for instance 1
Enter bot instance name: ender

###### Creating base config file
[sudo] password for pi:

###### moonraker-telegram-bot-ender.service instance created!

###### Launching moonraker-telegram-bot instance ...

###### Telegram bot instance name selection for instance 2
Enter bot instance name: tronxy

###### Creating base config file

###### moonraker-telegram-bot-tronxy.service instance created!

###### Launching moonraker-telegram-bot instance ...
```

Thank you for quick support!

aka13-404, 2022-05-05:
@morganfw Check out the new installer, it can do multiple instances on its own.

closed 2022-05-05 by aka13-404