nlef/moonraker-telegram-bot #134: [Bug]: No notification on the Smartphone?

gitbreiti, 2022-04-17:
### What happened and in what context?

I am not sure if it is a bug!

I have set the following:

[progress_notification]
percent: 20

The status is also updated, but I don't get any notification on my phone. I only see the update when I open the chat.

### What branch does this occur on?

master

### Reports only on latest master or development branches!

- [X] I have updated to latest development/master version before submitting the bug

nlef, 2022-04-20:
@gitbreiti 
a little workaround is commited in ad2d118
Could you please check it?

gitbreiti, 2022-04-21:
I have overwritten the 3 files. Is that correct? Now no image is sent.

gitbreiti, 2022-04-21:
@nlef 
I have now overwritten all the files. Now a new notification comes: Status has been updated...

When restarting the bot I get the following message: Unknown/bad items in [telegram_ui] section: status_single_message:true

If I delete the section everything works fine! :)

gitbreiti, 2022-04-22:
After a reboot everything worked.

Closed.

closed 2022-04-22 by gitbreiti