nlef/moonraker-telegram-bot #135: [Bug]: no response from /status

Fuckingnameless, 2022-04-21:
### What happened and in what context?

it shows uploading photo on telegram but doesn't send

### If needed, paste the relevant bot log contents here.

2022-04-21 01:50:48,006 - __main__ - ERROR - Exception while handling an update:
Traceback (most recent call last):
  File "/home/pi/moonraker-telegram-bot-env/lib/python3.7/site-packages/telegram/vendor/ptb_urllib3/urllib3/connectionpool.py", line 617, in urlopen
    chunked=chunked)
  File "/home/pi/moonraker-telegram-bot-env/lib/python3.7/site-packages/telegram/vendor/ptb_urllib3/urllib3/connectionpool.py", line 390, in _make_request
    conn.request(method, url, **httplib_request_kw)
  File "/usr/lib/python3.7/http/client.py", line 1260, in request
    self._send_request(method, url, body, headers, encode_chunked)
  File "/usr/lib/python3.7/http/client.py", line 1306, in _send_request
    self.endheaders(body, encode_chunked=encode_chunked)
  File "/usr/lib/python3.7/http/client.py", line 1255, in endheaders
    self._send_output(message_body, encode_chunked=encode_chunked)
  File "/usr/lib/python3.7/http/client.py", line 1069, in _send_output
    self.send(chunk)
  File "/usr/lib/python3.7/http/client.py", line 991, in send
    self.sock.sendall(data)
  File "/usr/lib/python3.7/ssl.py", line 1015, in sendall
    v = self.send(byte_view[count:])
  File "/usr/lib/python3.7/ssl.py", line 984, in send
    return self._sslobj.write(data)
socket.timeout: The write operation timed out

During handling of the above exception, another exception occurred:

Traceback (most recent call last):
  File "/home/pi/moonraker-telegram-bot-env/lib/python3.7/site-packages/telegram/utils/request.py", line 259, in _request_wrapper
    resp = self._con_pool.request(*args, **kwargs)
  File "/home/pi/moonraker-telegram-bot-env/lib/python3.7/site-packages/telegram/vendor/ptb_urllib3/urllib3/request.py", line 70, in request
    **urlopen_kw)
  File "/home/pi/moonraker-telegram-bot-env/lib/python3.7/site-packages/telegram/vendor/ptb_urllib3/urllib3/request.py", line 148, in request_encode_body
    return self.urlopen(method, url, **extra_kw)
  File "/home/pi/moonraker-telegram-bot-env/lib/python3.7/site-packages/telegram/vendor/ptb_urllib3/urllib3/poolmanager.py", line 244, in urlopen
    response = conn.urlopen(method, u.request_uri, **kw)
  File "/home/pi/moonraker-telegram-bot-env/lib/python3.7/site-packages/telegram/vendor/ptb_urllib3/urllib3/connectionpool.py", line 666, in urlopen
    _stacktrace=sys.exc_info()[2])
  File "/home/pi/moonraker-telegram-bot-env/lib/python3.7/site-packages/telegram/vendor/ptb_urllib3/urllib3/util/retry.py", line 347, in increment
    raise six.reraise(type(error), error, _stacktrace)
  File "/home/pi/moonraker-telegram-bot-env/lib/python3.7/site-packages/telegram/vendor/ptb_urllib3/urllib3/packages/six.py", line 685, in reraise
    raise value.with_traceback(tb)
  File "/home/pi/moonraker-telegram-bot-env/lib/python3.7/site-packages/telegram/vendor/ptb_urllib3/urllib3/connectionpool.py", line 617, in urlopen
    chunked=chunked)
  File "/home/pi/moonraker-telegram-bot-env/lib/python3.7/site-packages/telegram/vendor/ptb_urllib3/urllib3/connectionpool.py", line 390, in _make_request
    conn.request(method, url, **httplib_request_kw)
  File "/usr/lib/python3.7/http/client.py", line 1260, in request
    self._send_request(method, url, body, headers, encode_chunked)
  File "/usr/lib/python3.7/http/client.py", line 1306, in _send_request
    self.endheaders(body, encode_chunked=encode_chunked)
  File "/usr/lib/python3.7/http/client.py", line 1255, in endheaders
    self._send_output(message_body, encode_chunked=encode_chunked)
  File "/usr/lib/python3.7/http/client.py", line 1069, in _send_output
    self.send(chunk)
  File "/usr/lib/python3.7/http/client.py", line 991, in send
    self.sock.sendall(data)
  File "/usr/lib/python3.7/ssl.py", line 1015, in sendall
    v = self.send(byte_view[count:])
  File "/usr/lib/python3.7/ssl.py", line 984, in send
    return self._sslobj.write(data)
telegram.vendor.ptb_urllib3.urllib3.exceptions.ProtocolError: ('Connection aborted.', timeout('The write operation timed out'))

The above exception was the direct cause of the following exception:

Traceback (most recent call last):
  File "/home/pi/moonraker-telegram-bot-env/lib/python3.7/site-packages/telegram/ext/utils/promise.py", line 96, in run
    self._result = self.pooled_function(*self.args, **self.kwargs)
[... 87 lines cut ...]
  File "/home/pi/moonraker-telegram-bot/bot/notifications.py", line 178, in _notify
    disable_notification=silent,
  File "/home/pi/moonraker-telegram-bot-env/lib/python3.7/site-packages/telegram/bot.py", line 130, in decorator
    result = func(*args, **kwargs)
  File "/home/pi/moonraker-telegram-bot-env/lib/python3.7/site-packages/telegram/bot.py", line 738, in send_photo
    protect_content=protect_content,
  File "/home/pi/moonraker-telegram-bot-env/lib/python3.7/site-packages/telegram/ext/extbot.py", line 209, in _message
    protect_content=protect_content,
  File "/home/pi/moonraker-telegram-bot-env/lib/python3.7/site-packages/telegram/bot.py", line 336, in _message
    result = self._post(endpoint, data, timeout=timeout, api_kwargs=api_kwargs)
  File "/home/pi/moonraker-telegram-bot-env/lib/python3.7/site-packages/telegram/bot.py", line 296, in _post
    f'{self.base_url}/{endpoint}', data=data, timeout=effective_timeout
  File "/home/pi/moonraker-telegram-bot-env/lib/python3.7/site-packages/telegram/utils/request.py", line 359, in post
    result = self._request_wrapper('POST', url, fields=data, **urlopen_kwargs)
  File "/home/pi/moonraker-telegram-bot-env/lib/python3.7/site-packages/telegram/utils/request.py", line 265, in _request_wrapper
    raise NetworkError(f'urllib3 HTTPError {error}') from error
telegram.error.NetworkError: urllib3 HTTPError ('Connection aborted.', timeout('The write operation timed out'))

### What branch does this occur on?

master

### Reports only on latest master or development branches!

- [X] I have updated to latest development/master version before submitting the bug

Fuckingnameless, 2022-04-21:
same problem on develpment branch for ME

aka13-404, 2022-05-05:
@Fuckingnameless Are you still getting the issue? If so, would you please attach the whole log as a file?

closed 2022-06-02 by aka13-404