nlef/moonraker-telegram-bot #142: Status Message heater_fans

TodWulff, 2022-05-01:
### What would you like to see added?

I have some water cooling stuff on my Voron v2.4.  This upgrade includes:
- an in-line h2o temp sensor (status message good for this sensor https://i.imgur.com/QhMOJzq.png )
- PWM driven Coolant Pump with Tach Feedback (heater fan)
- PWM driven Radiator Fan with Tach Feedback (heater fan), and 
- Hall Effect Flow Meter (set up as a non-pwm heater fan) with Tach Feedback

I am requesting that y'all consider adding a `status_message_heater_fans` class of status message elements, such that these additional devices can be included in a status message (pwm val, tach rate, etc.), please.  https://i.imgur.com/eYwYVdl.png

Thank you, in advance.

~MHz

TodWulff, 2022-05-01:
I'll see if I can possibly push a telegram message periodically (via delayed_gcode) so as to at least have that information in the telegram clients.

nlef, 2022-05-01:
> I'll see if I can possibly push a telegram message periodically (via delayed_gcode) so as to at least have that information in the telegram clients.

take a look at [progress_notification] time parameter. this option allows you to send messages at a specified frequency

TodWulff, 2022-05-01:
Thank you.  Initial pass at delayed_gcode worked: https://i.imgur.com/v1pqDDY.png

I'll see if I can concatenate multiple lines into a single message and also take a peek at [progress_notification].

Thanks again!

~MHz

aka13-404, 2022-05-02:
@TodWulff Would you kindly check out the latest dev? It's not as pretty as it will be, but you should now see pwm and rpm in the status message. We sadly have 1 halfdead fan which does not display rpm properly, so it is hard to tell how well it works.

I would appreciate some feedback, since it will go into the next release.

TodWulff, 2022-05-02:
> from **Long-Term suggestions** to **In Progress** in [main de](/nlef/moonraker-telegram-bot/projects/1#card-81347163)

Will do.  I'm about 20 min from a print completing, so this is a good time.  I'd appreciate some hand-holding on getting an upgrade applied.   Here is my current update manager settings for the bot.  What do I need to change?:  https://i.imgur.com/rrYbCtP.png

```
[update_manager client moonraker-telegram-bot]
type: git_repo
path: ~/moonraker-telegram-bot
origin: https://github.com/nlef/moonraker-telegram-bot.git
env: ~/moonraker-telegram-bot-env/bin/python
requirements: scripts/requirements.txt
install_script: scripts/install.sh
```
Please let me know.  TIA.

~MHz

TodWulff, 2022-05-03:
Much progress!

https://i.imgur.com/7VmUvFb.png

https://i.imgur.com/If9muvt.png

Seems white space could be added between numerical value and 'RPM' label
Would suggest to have 'RPM: ' label before numerical value (note added colon)
Seems some scaling of the numerical value is in order - in looking at the RPM info in MS, it presents that the need to divide by 100 exists, casting result as an int (presenting fractional RPM is probably of very little value).?.

And, while y'all are in the code, can you see if you can nix the addition of white space to capitalized entity names, please? :)

aka13-404, 2022-05-03:
Thank you for the testing and the quick reply. We are nearing the release, and @nlef does not want to squeeze a whole status-message rework into it. The divisor is fixed, the space before RPMs as well, and regarding the general look and feel we will work in the next release cycle.

I'll tag you in the coming corresponding tickets, so you can add your ideas/visions.

TodWulff, 2022-05-04:
Outstanding.  Thank you.  This FR can be closed as fulfilled/implemented.

closed 2022-05-05 by aka13-404