nlef/moonraker-telegram-bot #148: [Bug]: FPS error without cam

shavrin777, 2022-05-08:
### What happened and in what context?

Without cam section in config i get 
> Printer online
> Config errors in section [camera]:
>   Option 'fps: 0': value is not above 0
> Please correct the configuration according to the wiki (https://github.com/nlef/moonraker-telegram-bot/wiki/Sample-config)

at startup

### What branch does this occur on?

master

### Reports only on latest master or development branches!

- [X] I have updated to latest development/master version before submitting the bug

aka13-404, 2022-05-08:
Disabling camera section should disable parameter checks.
Functionality needs to be tested again.

aka13-404, 2022-06-02:
Has been fixed on dev