nlef/moonraker-telegram-bot #149: [Bug]: openCV queue problem

ppcd9220, 2022-05-09:
### What happened and in what context?

When timelapse by layer height activated, and I issue /video or /status command, sometimes bot produce "No video" (nothing is sent) and "no video" in status.
Looking at logs it loos like there is no queue management with timelapse task (save pic) and sent to telegram.
Looking at forums I see comments that such bug could happen when you try to open camera twice or more times.
There probably should be some queue organized or at lease busy flag.

As a result: No video, no photo in status, and timelapse images are 10kb (empty) or corrupted.
After some time (probably there is a timeout somewhere) without additional concurrent requests it starts working again.

Latest Master release.
Orange Pi Zero 512
Ubuntu Focal.

### If needed, paste the relevant bot log contents here.

2022-05-09 21:13:47,479 - __main__ - ERROR - Exception while handling an update:
Traceback (most recent call last):
  File "/home/pi/moonraker-telegram-bot-env/lib/python3.8/site-packages/telegram/ext/dispatcher.py", line 555, in process_update
    handler.handle_update(update, self, check, context)
  File "/home/pi/moonraker-telegram-bot-env/lib/python3.8/site-packages/telegram/ext/handler.py", line 198, in handle_update
    return self.callback(update, context)
  File "/home/pi/moonraker-telegram-bot/bot/main.py", line 212, in get_video
    with cameraWrap.take_video_generator() as (video_bio, thumb_bio, width, height):
  File "/usr/lib/python3.8/contextlib.py", line 113, in __enter__
    return next(self.gen)
  File "/home/pi/moonraker-telegram-bot/bot/camera.py", line 302, in take_video_generator
    (video_bio, thumb_bio, width, height) = self.take_video()
  File "/home/pi/moonraker-telegram-bot/bot/camera.py", line 42, in wrapper
    result = func(self, *args, **kwargs)
  File "/home/pi/moonraker-telegram-bot/bot/camera.py", line 364, in take_video
    height, width, channels = frame.shape
AttributeError: 'NoneType' object has no attribute 'shape'
2022-05-09 21:19:33,312 - camera - WARNING - Reading video frames queue exception 
2022-05-09 21:19:38,294 - camera - WARNING - Writing video frames queue exception <built-in method with_traceback of Full object at 0xa5299208>
2022-05-09 22:20:22,918 - __main__ - ERROR - Exception while handling an update:
Traceback (most recent call last):
  File "/home/pi/moonraker-telegram-bot-env/lib/python3.8/site-packages/telegram/ext/dispatcher.py", line 555, in process_update
    handler.handle_update(update, self, check, context)
  File "/home/pi/moonraker-telegram-bot-env/lib/python3.8/site-packages/telegram/ext/handler.py", line 198, in handle_update
    return self.callback(update, context)
  File "/home/pi/moonraker-telegram-bot/bot/main.py", line 212, in get_video
    with cameraWrap.take_video_generator() as (video_bio, thumb_bio, width, height):
  File "/usr/lib/python3.8/contextlib.py", line 113, in __enter__
    return next(self.gen)
  File "/home/pi/moonraker-telegram-bot/bot/camera.py", line 302, in take_video_generator
    (video_bio, thumb_bio, width, height) = self.take_video()
  File "/home/pi/moonraker-telegram-bot/bot/camera.py", line 42, in wrapper
    result = func(self, *args, **kwargs)
  File "/home/pi/moonraker-telegram-bot/bot/camera.py", line 364, in take_video
    height, width, channels = frame.shape
AttributeError: 'NoneType' object has no attribute 'shape'
2022-05-09 22:22:26,001 - __main__ - ERROR - Exception while handling an update:
Traceback (most recent call last):
  File "/home/pi/moonraker-telegram-bot-env/lib/python3.8/site-packages/telegram/ext/dispatcher.py", line 555, in process_update
    handler.handle_update(update, self, check, context)
  File "/home/pi/moonraker-telegram-bot-env/lib/python3.8/site-packages/telegram/ext/handler.py", line 198, in handle_update
    return self.callback(update, context)
  File "/home/pi/moonraker-telegram-bot/bot/main.py", line 212, in get_video
    with cameraWrap.take_video_generator() as (video_bio, thumb_bio, width, height):
  File "/usr/lib/python3.8/contextlib.py", line 113, in __enter__
    return next(self.gen)
  File "/home/pi/moonraker-telegram-bot/bot/camera.py", line 302, in take_video_generator
    (video_bio, thumb_bio, width, height) = self.take_video()
  File "/home/pi/moonraker-telegram-bot/bot/camera.py", line 42, in wrapper
    result = func(self, *args, **kwargs)
  File "/home/pi/moonraker-telegram-bot/bot/camera.py", line 364, in take_video
    height, width, channels = frame.shape
AttributeError: 'NoneType' object has no attribute 'shape'
2022-05-09 23:02:06,545 - __main__ - ERROR - Exception while handling an update:
[... 196 lines cut ...]
May 09 19:15:38 orangepizero python[988]: 2022-05-09 19:15:38,506 - apscheduler.executors.default - INFO - Job "Notifier._send_print_start_info (trigger: date[2022-05-09 19:15:37 MSK], next run at: 2022-05-09 19:15:37 MSK)" executed successfully
May 09 19:15:37 orangepizero python[988]: 2022-05-09 19:15:37,220 - apscheduler.executors.default - INFO - Running job "Notifier._send_print_start_info (trigger: date[2022-05-09 19:15:37 MSK], next run at: 2022-05-09 19:15:37 MSK)" (scheduled at 2022-05-09 19:15:37.216325+03:00)
May 09 19:15:37 orangepizero python[988]: 2022-05-09 19:15:37,219 - apscheduler.scheduler - INFO - Removed job 58dcd1c482e0488fac4fc66102d2aec8
May 09 19:15:37 orangepizero python[988]: 2022-05-09 19:15:37,217 - apscheduler.scheduler - INFO - Added job "Notifier._send_print_start_info" to job store "default"
May 09 19:14:53 orangepizero python[988]: 2022-05-09 19:14:53,050 - apscheduler.scheduler - INFO - Removed job ws_reschedule
May 09 19:14:53 orangepizero python[988]: 2022-05-09 19:14:53,032 - apscheduler.executors.default - INFO - Job "reshedule (trigger: interval[0:00:02], next run at: 2022-05-09 19:14:55 MSK)" executed successfully
May 09 19:14:53 orangepizero python[988]: 2022-05-09 19:14:53,029 - apscheduler.executors.default - INFO - Running job "reshedule (trigger: interval[0:00:02], next run at: 2022-05-09 19:14:55 MSK)" (scheduled at 2022-05-09 19:14:53.024788+03:00)
May 09 19:14:51 orangepizero python[988]: 2022-05-09 19:14:51,026 - apscheduler.scheduler - INFO - Added job "reshedule" to job store "default"
May 09 19:14:49 orangepizero python[988]: 2022-05-09 19:14:49,714 - apscheduler.executors.default - INFO - Job "reshedule (trigger: interval[0:00:02], next run at: 2022-05-09 19:14:51 MSK)" executed successfully
May 09 19:14:49 orangepizero python[988]: 2022-05-09 19:14:49,710 - apscheduler.executors.default - INFO - Running job "reshedule (trigger: interval[0:00:02], next run at: 2022-05-09 19:14:49 MSK)" (scheduled at 2022-05-09 19:14:49.707010+03:00)
May 09 19:14:47 orangepizero python[988]: 2022-05-09 19:14:47,708 - apscheduler.scheduler - INFO - Added job "reshedule" to job store "default"
May 09 19:14:47 orangepizero python[988]: 2022-05-09 19:14:47,701 - __main__ - WARNING - klippy disconnect detected with message: notify_klippy_disconnected
May 09 18:21:03 orangepizero python[988]: 2022-05-09 18:21:03,161 - apscheduler.scheduler - INFO - Removed job ws_reschedule
May 09 18:21:03 orangepizero python[988]: 2022-05-09 18:21:03,130 - apscheduler.scheduler - INFO - Added job "reshedule" to job store "default"
May 09 18:21:02 orangepizero python[988]: 2022-05-09 18:21:02,445 - apscheduler.scheduler - INFO - Scheduler started
May 09 18:21:02 orangepizero python[988]: 2022-05-09 18:21:02,127 - apscheduler.scheduler - INFO - Scheduler started
May 09 18:20:42 orangepizero systemd[1]: Started Starts Moonraker Telegram Bot on startup.

### What branch does this occur on?

master

### Reports only on latest master or development branches!

- [X] I have updated to latest development/master version before submitting the bug

ppcd9220, 2022-05-09:
Add on.
Video source is local network stream form IP camera:
rtsp://192.168.1.92/live/ch00_1
Codec: H.264

aka13-404, 2022-05-11:
This is rather strange, and the bot does all the necessary checks which you mention already. It needs further testing, what specifically causes this issue with your stream. I'll get back to you when I have more information.
In the meantime, you could try a different format besides rtsp if your camera supports it.

closed 2022-06-12 by aka13-404