nlef/moonraker-telegram-bot #154: [Bug]: Repeat Messages

4noxx, 2022-05-17:
### What happened and in what context?

i get all one or two seconds this message here:

Status has been updated
This message will be deleted

[image]

### If needed, paste the relevant bot log contents here.

The above exception was the direct cause of the following exception:

Traceback (most recent call last):
  File "/home/klipper/moonraker-telegram-bot-env/lib/python3.9/site-packages/apscheduler/executors/base.py", line 125, in run_job
    retval = job.func(*job.args, **job.kwargs)
  File "/home/klipper/moonraker-telegram-bot/bot/notifications.py", line 351, in _notify_by_time
    self._schedule_notification()
  File "/home/klipper/moonraker-telegram-bot/bot/notifications.py", line 324, in _schedule_notification
    self._notify(mess, self._silent_progress, self._group_only)
  File "/home/klipper/moonraker-telegram-bot/bot/notifications.py", line 173, in _notify
    self._bot.send_chat_action(chat_id=self._chat_id, action=ChatAction.UPLOAD_PHOTO)
  File "/home/klipper/moonraker-telegram-bot-env/lib/python3.9/site-packages/telegram/bot.py", line 130, in decorator
    result = func(*args, **kwargs)
  File "/home/klipper/moonraker-telegram-bot-env/lib/python3.9/site-packages/telegram/bot.py", line 2140, in send_chat_action
    result = self._post('sendChatAction', data, timeout=timeout, api_kwargs=api_kwargs)
  File "/home/klipper/moonraker-telegram-bot-env/lib/python3.9/site-packages/telegram/bot.py", line 295, in _post
    return self.request.post(
  File "/home/klipper/moonraker-telegram-bot-env/lib/python3.9/site-packages/telegram/utils/request.py", line 361, in post
    result = self._request_wrapper(
  File "/home/klipper/moonraker-telegram-bot-env/lib/python3.9/site-packages/telegram/utils/request.py", line 265, in _request_wrapper
    raise NetworkError(f'urllib3 HTTPError {error}') from error
telegram.error.NetworkError: urllib3 HTTPError HTTPSConnectionPool(host='api.telegram.org', port=443): Max retries exceeded with url: /botxxxxxxxxxxxxxxxx/sendChatAction (Caused by NewConnectionError('<telegram.vendor.ptb_urllib3.urllib3.connection.VerifiedHTTPSConnection object at 0xa0cd57d8>: Failed to establish a new connection: [Errno -3] Temporary failure in name resolution'))

### If needed, paste the relevant bot journal contents here.

klipper@raspberrypi:~ $ sudo journalctl -r -u moonraker-telegram-bot
-- Journal begins at Sun 2022-04-17 15:17:01 CEST, ends at Tue 2022-05-17 12:53>
Mai 17 11:13:50 raspberrypi systemd[1]: moonraker-telegram-bot.service: Consume>
Mai 17 11:13:50 raspberrypi systemd[1]: Stopped Starts Moonraker Telegram Bot o>
Mai 17 11:13:50 raspberrypi systemd[1]: moonraker-telegram-bot.service: Succeed>
Mai 17 11:13:50 raspberrypi systemd[1]: Stopping Starts Moonraker Telegram Bot >
Mai 17 11:13:48 raspberrypi python[834]: 2022-05-17 11:13:48,535 - apscheduler.>
Mai 17 11:13:47 raspberrypi python[834]: [mjpeg @ 0x9eda3630] unable to decode >
Mai 17 11:13:47 raspberrypi python[834]: [mjpeg @ 0x9ed82360] unable to decode >
Mai 17 11:13:47 raspberrypi python[834]: [mjpeg @ 0x9c743330] unable to decode >
Mai 17 11:13:47 raspberrypi python[834]: [mjpeg @ 0x9f785fd0] unable to decode >
Mai 17 11:13:47 raspberrypi python[834]: 2022-05-17 11:13:47,293 - apscheduler.>
Mai 17 11:13:47 raspberrypi python[834]: 2022-05-17 11:13:47,292 - apscheduler.>
Mai 17 11:13:47 raspberrypi python[834]: 2022-05-17 11:13:47,290 - apscheduler.>
Mai 17 11:13:43 raspberrypi python[834]: 2022-05-17 11:13:43,591 - apscheduler.>
Mai 17 11:13:42 raspberrypi python[834]: [mjpeg @ 0xa0b06150] unable to decode >
Mai 17 11:13:42 raspberrypi python[834]: [mjpeg @ 0xa0b0a6f0] unable to decode >
Mai 17 11:13:42 raspberrypi python[834]: [mjpeg @ 0xa0100bb0] unable to decode >
Mai 17 11:13:42 raspberrypi python[834]: [mjpeg @ 0xa010ab40] unable to decode >
Mai 17 11:13:42 raspberrypi python[834]: 2022-05-17 11:13:42,293 - apscheduler.>
Mai 17 11:13:42 raspberrypi python[834]: 2022-05-17 11:13:42,292 - apscheduler.>
Mai 17 11:13:42 raspberrypi python[834]: 2022-05-17 11:13:42,290 - apscheduler.>
Mai 17 11:13:41 raspberrypi python[834]: [mjpeg @ 0x9ed82360] unable to decode >
Mai 17 11:13:41 raspberrypi python[834]: [mjpeg @ 0x9ed82780] unable to decode >
lines 1-23
-- Journal begins at Sun 2022-04-17 15:17:01 CEST, ends at Tue 2022-05-17 12:53:54 CEST. --
Mai 17 11:13:50 raspberrypi systemd[1]: moonraker-telegram-bot.service: Consumed 2min 3.751s CPU time.
Mai 17 11:13:50 raspberrypi systemd[1]: Stopped Starts Moonraker Telegram Bot on startup.
Mai 17 11:13:50 raspberrypi systemd[1]: moonraker-telegram-bot.service: Succeeded.
Mai 17 11:13:50 raspberrypi systemd[1]: Stopping Starts Moonraker Telegram Bot on startup...
Mai 17 11:13:48 raspberrypi python[834]: 2022-05-17 11:13:48,535 - apscheduler.executors.default - INFO - Job "Notifier._notify_by_time (trigger: interval[0:00:05], next run at:>
Mai 17 11:13:47 raspberrypi python[834]: [mjpeg @ 0x9eda3630] unable to decode APP fields: Invalid data found when processing input
Mai 17 11:13:47 raspberrypi python[834]: [mjpeg @ 0x9ed82360] unable to decode APP fields: Invalid data found when processing input
Mai 17 11:13:47 raspberrypi python[834]: [mjpeg @ 0x9c743330] unable to decode APP fields: Invalid data found when processing input
[... 21 lines cut ...]
Mai 17 11:13:47 raspberrypi python[834]: [mjpeg @ 0x9eda3630] unable to decode APP fields: Invalid data found when processing input
Mai 17 11:13:47 raspberrypi python[834]: [mjpeg @ 0x9ed82360] unable to decode APP fields: Invalid data found when processing input
Mai 17 11:13:47 raspberrypi python[834]: [mjpeg @ 0x9c743330] unable to decode APP fields: Invalid data found when processing input
Mai 17 11:13:47 raspberrypi python[834]: [mjpeg @ 0x9f785fd0] unable to decode APP fields: Invalid data found when processing input
Mai 17 11:13:47 raspberrypi python[834]: 2022-05-17 11:13:47,293 - apscheduler.executors.default - INFO - Job "Timelapse.take_lapse_photo (trigger: interval[0:00:05], next run at: 2022-05-17 11:13:52 CEST)" executed success>
Mai 17 11:13:47 raspberrypi python[834]: 2022-05-17 11:13:47,292 - apscheduler.executors.default - INFO - Running job "Timelapse.take_lapse_photo (trigger: interval[0:00:05], next run at: 2022-05-17 11:13:52 CEST)" (schedul>
Mai 17 11:13:47 raspberrypi python[834]: 2022-05-17 11:13:47,290 - apscheduler.executors.default - INFO - Running job "Notifier._notify_by_time (trigger: interval[0:00:05], next run at: 2022-05-17 11:13:52 CEST)" (scheduled>
Mai 17 11:13:43 raspberrypi python[834]: 2022-05-17 11:13:43,591 - apscheduler.executors.default - INFO - Job "Notifier._notify_by_time (trigger: interval[0:00:05], next run at: 2022-05-17 11:13:47 CEST)" executed successfu>
Mai 17 11:13:42 raspberrypi python[834]: [mjpeg @ 0xa0b06150] unable to decode APP fields: Invalid data found when processing input
Mai 17 11:13:42 raspberrypi python[834]: [mjpeg @ 0xa0b0a6f0] unable to decode APP fields: Invalid data found when processing input
Mai 17 11:13:42 raspberrypi python[834]: [mjpeg @ 0xa0100bb0] unable to decode APP fields: Invalid data found when processing input
Mai 17 11:13:42 raspberrypi python[834]: [mjpeg @ 0xa010ab40] unable to decode APP fields: Invalid data found when processing input
Mai 17 11:13:42 raspberrypi python[834]: 2022-05-17 11:13:42,293 - apscheduler.executors.default - INFO - Job "Timelapse.take_lapse_photo (trigger: interval[0:00:05], next run at: 2022-05-17 11:13:47 CEST)" executed success>
Mai 17 11:13:42 raspberrypi python[834]: 2022-05-17 11:13:42,292 - apscheduler.executors.default - INFO - Running job "Timelapse.take_lapse_photo (trigger: interval[0:00:05], next run at: 2022-05-17 11:13:47 CEST)" (schedul>
Mai 17 11:13:42 raspberrypi python[834]: 2022-05-17 11:13:42,290 - apscheduler.executors.default - INFO - Running job "Notifier._notify_by_time (trigger: interval[0:00:05], next run at: 2022-05-17 11:13:47 CEST)" (scheduled>
Mai 17 11:13:41 raspberrypi python[834]: [mjpeg @ 0x9ed82360] unable to decode APP fields: Invalid data found when processing input
Mai 17 11:13:41 raspberrypi python[834]: [mjpeg @ 0x9ed82780] unable to decode APP fields: Invalid data found when processing input

### What branch does this occur on?

master

### Reports only on latest master or development branches!

- [X] I have updated to latest development/master version before submitting the bug

aka13-404, 2022-05-17:
Well, you have two things going on: 
1. You update your status message every 1-2 seconds, if you get this message every 1-2 messages
2. You have not disabled notifications for status message updates 

You have two options - 
1.)reduce the interval in which you want the status updated
2.)Disable notifications for status updates with silent_progress: true

4noxx, 2022-05-17:
thx.... i dont whats the problem. after restart i get not repeat notifys (status). if i press STATUS, i get this:
[image]

every time

LOG
Traceback (most recent call last):
  File "/home/klipper/moonraker-telegram-bot-env/lib/python3.9/site-packages/apscheduler/executors/base.py", line 125, in run_job
    retval = job.func(*job.args, **job.kwargs)
  File "/home/klipper/moonraker-telegram-bot/bot/notifications.py", line 351, in _notify_by_time
    self._schedule_notification()
  File "/home/klipper/moonraker-telegram-bot/bot/notifications.py", line 324, in _schedule_notification
    self._notify(mess, self._silent_progress, self._group_only)
  File "/home/klipper/moonraker-telegram-bot/bot/notifications.py", line 173, in _notify
    self._bot.send_chat_action(chat_id=self._chat_id, action=ChatAction.UPLOAD_PHOTO)
  File "/home/klipper/moonraker-telegram-bot-env/lib/python3.9/site-packages/telegram/bot.py", line 130, in decorator
    result = func(*args, **kwargs)
  File "/home/klipper/moonraker-telegram-bot-env/lib/python3.9/site-packages/telegram/bot.py", line 2140, in send_chat_action
    result = self._post('sendChatAction', data, timeout=timeout, api_kwargs=api_kwargs)
  File "/home/klipper/moonraker-telegram-bot-env/lib/python3.9/site-packages/telegram/bot.py", line 295, in _post
    return self.request.post(
  File "/home/klipper/moonraker-telegram-bot-env/lib/python3.9/site-packages/telegram/utils/request.py", line 361, in post
    result = self._request_wrapper(
  File "/home/klipper/moonraker-telegram-bot-env/lib/python3.9/site-packages/telegram/utils/request.py", line 265, in _request_wrapper
    raise NetworkError(f'urllib3 HTTPError {error}') from error
telegram.error.NetworkError: urllib3 HTTPError HTTPSConnectionPool(host='api.telegram.org', port=443): Max retries exceeded with
url: /bot_xxxxxxxxxxxxx:YYYYYYYYYYYYYYYYYYYY/sendChatAction (Caused by NewConnectionError('<telegram.vendor.ptb_urllib3.urllib3.connection.VerifiedHTTPSConnection
object at 0xa0cd57d8>: Failed to establish a new connection: [Errno -3] Temporary failure in name resolution'))

aka13-404, 2022-05-17:
I already told you, what the problem is.

closed 2022-05-20 by aka13-404