nlef/moonraker-telegram-bot #156: Help with disabled_macros

teeedubb, 2022-05-19:
### What happened and in what context?

Hi,

I'm trying to hide some macros in the telegram ui but they are not being hidden.

The line in my config file:
[telegram_ui]
silent_progress: true
eta_source: file
disabled_macros: m300, /idle_timeout_macro, /filament_runout_pause, /end_gcode, /start_gcode, /shutdown, /light

(I have tried both with and without the leading slash)
But they are still visible:

[image]

I'm not sure what's wrong - does anyone have any pointers?

Thanks

### What branch does this occur on?

master

### Reports only on latest master or development branches!

- [X] I have updated to latest development/master version before submitting the bug

aka13-404, 2022-05-20:
Thank you for the report, I'll look into it and get back to you.

teeedubb, 2022-05-22:
After switching to the the dev branch with the above commit my custom macros are now disabled - Thanks!
One question though - should I be able to disable the inbuilt macros? I have added light and shutdown to the list and they still display in the telegram ui.

aka13-404, 2022-05-22:
That's an interesting thought. We never considered that you may want to disable the bots internal command autocompletion. So far, you are not be able to, but don't be dissuaded, we'll look intohow this might be done in the future.
I have openend a separate feature ticket, #157  you might want to subscribe so you will be notified when there is progress on that.

The issue with non-hideable-macros will roll out in the next release.

teeedubb, 2022-05-22:
Thanks for letting me know. My use case is that I have my own macros for lights and system shutdown so it would be nice to to be able to remove the builtin ones from the list to declutter it.

Thanks again for your assistance and work on this awesome addition to klipper 👍

closed 2022-06-02 by aka13-404