nlef/moonraker-telegram-bot #161: [Bug]: When starting printing, everything stops and restarts the system

rdlvm, 2022-05-27:
### What happened and in what context?

My printer works correctly, but after installing Telegram bot, just when starting a new print it stops and throws these errors, if I stop the Telegram Bot service everything works perfect, if I activate it, the same thing happens again, the system restarts right after printing starts.

[url=https://imgbb.com/][img]https://i.ibb.co/RbX69tN/photo-2022-05-24-16-40-46.jpg[/img][/url]

### What branch does this occur on?

master

### Reports only on latest master or development branches!

- [X] I have updated to latest development/master version before submitting the bug

aka13-404, 2022-06-01:
What system are you runnning on? To me it looks simply that your system is overloaded, and the bot is simply too much for it.

rdlvm, 2022-06-01:
> What system are you runnning on? To me it looks simply that your system is overloaded, and the bot is simply too much for it.

Hello, i use mainsailos, on a raspberry pi zero 2 w, i don't see overload when i use the performance monitor.

I have tried two clean installs and on both clean installs the same thing happens.

aka13-404, 2022-06-01:
The 2w is to weak, it only has 512MB of ram. It is not enough for operating the bot together with klipper. 
You simply overload the ram, and swap is not fast enough.

rdlvm, 2022-06-01:
> El 2w es débil, solo tiene 512 MB de ram. No es suficiente operar el bot junto con klipper. Simplemente sobrecarga la memoria RAM y el intercambio no es lo suficientemente rápido.

closed 2022-06-01 by rdlvm

aka13-404, 2022-06-02:
Even if it helps, that you send less updates, you will definitely run into memory issues and klipper stops. 
I mean it, your system is too weak, it can and will cause issues on longer/more complex prints

rdlvm, 2022-06-02:
> Even if it helps, that you send less updates, you will definitely run into memory issues and klipper stops.
> I mean it, your system is too weak, it can and will cause issues on longer/more complex prints

Understood, any alternative with the bot to be reliable and consume less resources?  maybe optimize it?

aka13-404, 2022-06-02:
You **LITERALLY HAVE 512MB RAM**
You need about 150MB for the system
About 200MB for klipper to run
About 100MB for moonraker. 

That is already 450mb just for a normal print with low resolution and minimal microstepping. You are lucky, that klipper functions at all on your system. You should avoid installing anything at all there. Any complex curve can bring your print to a halt even without the bot. 

There is nothing and will be nothing, that can fit in 50mb ram and do any meaningful interaction with moonraker and something else on top of it.