nlef/moonraker-telegram-bot #165: [Bug]: Service does not start

LionBit76, 2022-06-02:
### What happened and in what context?

After booting the printer, Telegram-Bot does not start. Only after a service restart does it work as it should. I have already reinstalled Telegram-Bot, but there is no improvement. The log after boot looks like this:

### If needed, paste the relevant bot journal contents here.

-- Logs begin at Thu 2022-06-02 02:32:24 CEST, end at Thu 2022-06-02 02:46:37 CEST. --
Jun 02 02:32:48 Voron python[672]:     stack_info=True,
Jun 02 02:32:48 Voron python[672]:   File "/home/pi/moonraker-telegram-bot/bot/main.py", line 54, in handle_exception
Jun 02 02:32:48 Voron python[672]: Stack (most recent call last):
Jun 02 02:32:48 Voron python[672]: telegram.error.BadRequest: Can't parse entities: character '!' is reserved and must be escaped with the preceding '\'
Jun 02 02:32:48 Voron python[672]:     raise BadRequest(message)
Jun 02 02:32:48 Voron python[672]:   File "/home/pi/moonraker-telegram-bot-env/lib/python3.7/site-packages/telegram/utils/request.py", line 279, in _request_wrapper
Jun 02 02:32:48 Voron python[672]:     **urlopen_kwargs,
Jun 02 02:32:48 Voron python[672]:   File "/home/pi/moonraker-telegram-bot-env/lib/python3.7/site-packages/telegram/utils/request.py", line 366, in post
Jun 02 02:32:48 Voron python[672]:     f'{self.base_url}/{endpoint}', data=data, timeout=effective_timeout
Jun 02 02:32:48 Voron python[672]:   File "/home/pi/moonraker-telegram-bot-env/lib/python3.7/site-packages/telegram/bot.py", line 296, in _post
Jun 02 02:32:48 Voron python[672]:     result = self._post(endpoint, data, timeout=timeout, api_kwargs=api_kwargs)
Jun 02 02:32:48 Voron python[672]:   File "/home/pi/moonraker-telegram-bot-env/lib/python3.7/site-packages/telegram/bot.py", line 336, in _message
Jun 02 02:32:48 Voron python[672]:     protect_content=protect_content,
Jun 02 02:32:48 Voron python[672]:   File "/home/pi/moonraker-telegram-bot-env/lib/python3.7/site-packages/telegram/ext/extbot.py", line 209, in _message
Jun 02 02:32:48 Voron python[672]:     protect_content=protect_content,
Jun 02 02:32:48 Voron python[672]:   File "/home/pi/moonraker-telegram-bot-env/lib/python3.7/site-packages/telegram/bot.py", line 531, in send_message
Jun 02 02:32:48 Voron python[672]:     result = func(*args, **kwargs)
Jun 02 02:32:48 Voron python[672]:   File "/home/pi/moonraker-telegram-bot-env/lib/python3.7/site-packages/telegram/bot.py", line 130, in decorator
Jun 02 02:32:48 Voron python[672]:     disable_notification=notifier.silent_status,
Jun 02 02:32:48 Voron python[672]:   File "/home/pi/moonraker-telegram-bot/bot/main.py", line 928, in greeting_message
Jun 02 02:32:48 Voron python[672]:     greeting_message()
Jun 02 02:32:48 Voron python[672]:   File "/home/pi/moonraker-telegram-bot/bot/main.py", line 1408, in <module>
Jun 02 02:32:48 Voron python[672]: Traceback (most recent call last):
Jun 02 02:32:48 Voron python[672]: 2022-06-02 02:32:48,397 - __main__ - ERROR - Uncaught exception
Jun 02 02:32:48 Voron python[672]: 2022-06-02 02:32:48,283 - apscheduler.scheduler - INFO - Scheduler started
Jun 02 02:32:48 Voron python[672]: 2022-06-02 02:32:48,168 - apscheduler.scheduler - INFO - Scheduler started
Jun 02 02:32:38 Voron systemd[1]: Started Starts Moonraker Telegram Bot on startup.

### What branch does this occur on?

master

### Reports only on latest master or development branches!

- [X] I have updated to latest development/master version before submitting the bug

aka13-404, 2022-06-02:
Can you attach the config or the full log pls? I have a feeling, you have a "!" somewhere in your macro/command names.

LionBit76, 2022-06-02:
My printer config is here:  [https://github.com/LionBit76/VORON-2.4-250](url)
the telegram.conf is attached:
[telegram.conf.txt](https://github.com/nlef/moonraker-telegram-bot/files/8822491/telegram.conf.txt)
Is this the log, you want?
[mtb.log](https://github.com/nlef/moonraker-telegram-bot/files/8822610/mtb.log)

nlef, 2022-06-03:
@LionBit76
Some workround published in dev branch. Could you test it?

LionBit76, 2022-06-03:
@nlef 
Oh, yes, it works! 
Thank you :)

LionBit76, 2022-06-03:
Now I have to make changes in the moonraker.conf, what should that look like?

nlef, 2022-06-04:
> Now I have to make changes in the moonraker.conf, what should that look like?

look in the bot log to find problematic macro names and correct them in klipper/moonraker conf files.

LionBit76, 2022-06-04:
> 

I mean the section for the updater :) 
My config in moonraker.conf is:

[update_manager client moonraker-telegram-bot]
type: git_repo
path: ~/moonraker-telegram-bot
origin: https://github.com/nlef/moonraker-telegram-bot.git
env: ~/moonraker-telegram-bot-env/bin/python
requirements: scripts/requirements.txt
install_script: scripts/install.sh

...but it need a change here...

nlef, 2022-06-05:
> I mean the section for the updater :)
> My config in moonraker.conf is:
>
> [update_manager client moonraker-telegram-bot]
> type: git_repo
> path: ~/moonraker-telegram-bot
> origin: [...]

Mqybe you should use 'primary_branch' parameter in update manager section for bot?
But be careful, development branch is not stable)

closed 2022-06-14 by aka13-404