nlef/moonraker-telegram-bot #166: [Bug]: Additional status message info not showing

Disttrack, 2022-06-05:
### What happened and in what context?

I have added several additional settings to the status message, but these are not displayed in the status message.

### If needed, paste the relevant bot log contents here.

[image]
[image]
[image]

### If needed, paste the relevant bot journal contents here.

[journalctl.log](https://github.com/nlef/moonraker-telegram-bot/files/8839905/journalctl.log)

### What branch does this occur on?

master

### Reports only on latest master or development branches!

- [X] I have updated to latest development/master version before submitting the bug

nlef, 2022-06-07:
@Disttrack 
I can't figure out why it doesn't work from the attached log. Сoud you provide the complete single print bot log?

Disttrack, 2022-06-07:
> @Disttrack I can't figure out why it doesn't work from the attached log. Сoud you provide the complete single print bot log?

I uninstalled the bot and reinstall It, now its working good. Thanks.

closed 2022-06-07 by Disttrack