nlef/moonraker-telegram-bot #169: [Bug]: High Resource Usage on RPI 3A

503Dev, 2022-06-10:
### What happened and in what context?

1. Install 3 instances of telegram bot via Kiauh
2. Configure each instance with an API key and chat ID

After doing so I noticed my RPI was staying at max usage on 1 core but more importantly, the memory usage went from 180MB to 900MB on average with the telegram bot script consuming 23% per instance.

I just wanted to report this as it does seem abnormal.  As a fellow dev I can't fathom how a Python script which interacts with the Telegram API would cause such high usage, this could be a bug with a package maybe?

### What branch does this occur on?

master

### Reports only on latest master or development branches!

- [X] I have updated to latest development/master version before submitting the bug

aka13-404, 2022-06-10:
There is nothing unusual in that kind of usage, and absolutely expected.
The bot centers around operations on the video stream, an so opencv2 libraries play a core part in it.
Those libraries alone need around 100-150MB per instance, and corresponding CPU power. 
Interacting with moonraker is not free ram-wise, neither are telegram libraries. 
200-250MB are absolutely normal for normal operations for an instance, and while building lapses or sufficiently long buffers for videos, can easily exceed 1-2 gigs. 

The amount of users that we have who do not use the camera is very little, so there was no incentive so far to make them separate from the bot. The idea is in the backlog, but there is definitely no ETA on when, if at all, no-camera-mode will become more lightweight compared to the default install.

If you were intending to run 3 instances, with real-time image operations while printing on an Rpi 3A, so help you God.

503Dev, 2022-06-11:
aka13-404 thank you for the great reply.  Indeed makes sense, I had no factored in camera operations as I was not concerned with that but I have a very unique use case.  I think it's fair to say it would be fairly pointless to decouple the logic as the vast majority of use cases for this would be centered around cameras and time lapse retrieval.  I will fork and work on my own logic to negate that but again that is entirely due to my weird situation, the overall bot seems excellent and granted the video logic is integral the memory usage seems totally reasonable.

aka13-404, 2022-06-11:
Appreciate the warm words and the understanding. I will mention you, when/if camera libraries can be turned off in stock.