nlef/moonraker-telegram-bot #173: Telegram bot: Announcements and config breaking changes

aka13-404, 2022-06-12:
Hello everyone!

We have finally implemented announcements as well, and intend to use them to warn you in advance about config breaking changes in our releases. I hope that this will be a good tool, and not a simple annoyance. This could also be a great place to discuss problems and issues with our releases.

We just have published a new release, with small QoL additions and bugfixes.
Sadly, we had to break `disabled_macros` and `show_hidden_macros`, but this should be an easy 2-minute fix.

The whoel changelog is available under the releases page:
https://github.com/nlef/moonraker-telegram-bot/releases/tag/v.1.4.4

Nkawu, 2022-06-15:
Typo in the URL: https://github.com/nlef/moonraker-telegram-bot/releases/tag/v1.4.4

aka13-404, 2022-06-16:
> Typo in the URL: https://github.com/nlef/moonraker-telegram-bot/releases/tag/v1.4.4

Actually, it was much worse, it was a typo in the release name :D 
Thanks, updated the link.

federico9292, 2022-06-26:
[image]
typo in the question "Execute **marco** CANCEL_PRINT?"

Nkawu, 2022-06-26:
> typo in the question "Execute **marco** CANCEL_PRINT?"

I think we should spare Marco's life!

Fuckingnameless, 2022-06-27:
log path is broken?

aka13-404, 2022-06-27:
> log path is broken?

Please elaborate, how it is broken?

> typo in the question "Execute marco CANCEL_PRINT?"
Fixed for the next version, thank you :D

federico9292, 2022-06-29:
can I just add that timing is not considering timezones? finish at 2022-06-29   01:56 but it's 02:30....! it will finish at 02:56...

federico9292, 2022-06-29:
> It's propably you, who is not considering timezones ;) Did you check the
> time on your printer?

what do you mean with time on the printer? I mean, Fluidd said correctly
that the print would have finished at 2:56

aka13-404, 2022-06-29:
> what do you mean with time on the printer? I mean, Fluidd said correctly that the print would have finished at 2:56 Il giorno mer 29 giu 2022 alle 11:20 Alexander Krasnov < ***@***.***> ha scritto:
> […](#)

I mean, check the time on your printer by connecting via ssh please.

closed 2022-08-22 by aka13-404