nlef/moonraker-telegram-bot #174: Storing sensive data outside of config file

shcad, 2022-06-12:
### What would you like to see added?

I use an automation to store all my config files from klipper on github. Sensitive information storage in config file (tokens and chatid)  seams to be a vulnerability for me. 

Would be better if that sensitive data could be stored outside config file directory and put in config with "an envelope" (like the "secrets" in moonraker.conf).

aka13-404, 2022-06-12:
You can use native git mechanisms to do exactly that. 
Using clean and smudge is a good practice, you can check out my printer repo on a general feel on how it is done.

What you request is definitely desirable, but not on the to-do list in the foreseeable future, I am afraid.
This is a very good suggestion ticket to track progress on that feature.

aka13-404, 2022-08-22:
Working on dev

closed 2022-09-10 by aka13-404