nlef/moonraker-telegram-bot #176: Telegram Message annoyance

Minsekt, 2022-06-13:
Im having a problem where the bot sends me [image] this every 5sec. However my progress report is set to 10sec and I've set silent progress to true but it still happens. Happens after the latest update

aka13-404, 2022-06-13:
Please attach a log when reporting issues.

Sirnut, 2022-06-13:
This issue started happening to me just now, but I haven't actually updated in a while

[journalctl.txt](https://github.com/nlef/moonraker-telegram-bot/files/8894016/journalctl.txt)
[telegram.log](https://github.com/nlef/moonraker-telegram-bot/files/8894017/telegram.log)

aka13-404, 2022-06-14:
@Sirnut I deleted your log, since the last version of the bot dumped the token/chatid into the log. If you update, it should be safe again to post logs. But as far as I could see, you had simple internet issues, what does it have to do with notifications?

aka13-404, 2022-06-14:
@Minsekt 
I have forgotten to updaate the release notes. 

Progress update messages configuration has now been **split into two parts:** `progress_update_message: true` enables the sending of the update message, and `silent_progress: false` to set notification type to silent.

Sorry for the confusion. The thing that you get an update every 5 seconds though, is your configuration.

Sirnut, 2022-06-14:
@Minsekt Oops, thank you for that. I'm not sure what was causing the notification issues, as yesterday I wasn't having internet issues in the morning (on my desktop at least)

Now that you mention it, I did reboot my router later in the day and haven't had the issue since, but I also haven't printed anything

Minsekt, 2022-06-14:
@aka13-404 it started happening for me at the last but one update. I then used silent_progress: true as a workaround which got the job done. Now I can't seem to fix it [image]

When using progress_update_message I get an error

Minsekt, 2022-06-14:
Removed the bot and did a clean install, now it's working.

Sirnut, 2022-06-14:
@Minsekt I'll give that a shot too

aka13-404, 2022-06-14:
@Minsekt We had a bug with the `progress_update_message:´ option. I have just updated the wiki, and nlef rolled out a fix for the config checker. When setting `progress_update_message: false´ in the gui section, it should now work as expected, at least it does so for me. Please let me know how it works out.

Minsekt, 2022-06-15:
> @Minsekt We had a bug with the `progress_update_message:´ option. I have just updated the wiki, and nlef rolled out a fix for the config checker. When setting `progress_update_message: false´ in the [...]

yes its working now. thanks!

closed 2022-06-15 by Minsekt

Sirnut, 2022-06-17:
So after a fresh install today with "silent_status: true" I still get this message

This wasn't happening for months. Why would it all of a sudden have started without prior to any updates

Sirnut, 2022-06-17:
I am unsure whether this plays a role, but I manually install my bot and then add it to the moonraker update manager. I back my klipper_config up to a public repository, so I cannot keep the telegram.conf inside there for security reasons

aka13-404, 2022-06-17:
Because this has nothing to do with getting the message as per this ticket, changelog or wiki entries.