nlef/moonraker-telegram-bot #177: [Bug]: No progress notification

tekgamerpt, 2022-06-15:
### What happened and in what context?

Hello
Im new on klipper but i use telegram bots for several things.
At the beginning everything worked fine but now i dont have any start or ending messages because progress is always at 0,00%
Bot is alive and respondes to commands but something is not going well with this progress percentage. It seems that it does not have acess to this type of info.
For sure this is not a bug of the bot itself but something wrong on my side, some config that im missing.
Can anyone help me?

I have installed everythig using kiauh

### If needed, paste the relevant bot log contents here.

Current Monraker telegram bot config
[bot]
server = localhost
bot_token = xxxxxxxxxxxxxxxxxxxx
chat_id = xxxxxxxxxxxxx

[camera]
host = http://1xxxxxxxx

[progress_notification]
percent = 50
height = 0
time = 0

[timelapse]
cleanup = true
height = 0.2
time = 0
target_fps = 15
send_finished_lapse = true

[telegram_ui]
eta_source = slicer
buttons = [status,pause,cancel,resume],[files,emergency,macros,shutdown],[photo,video,macros]
require_confirmation_macro = true
silent_progress = false
show_private_macros = false
status_message_content = progress, height, filament_length, filament_weight, print_duration, eta, finish_time, m117_status, tgnotify_status, last_update_time
status_message_m117_update = true
status_message_sensors = mcu, host
status_message_heaters = extruder, heater_bed

2022-06-15 14:30:20,050 - __main__ - ERROR - Uncaught exception
Traceback (most recent call last):
  File "/home/pi/moonraker-telegram-bot/bot/main.py", line 1439, in <module>
    greeting_message()
  File "/home/pi/moonraker-telegram-bot/bot/main.py", line 976, in greeting_message
    bot_updater.bot.set_my_commands(commands=prepare_commands_list(klippy.macros, configWrap.telegram_ui.include_macros_in_command_list))
  File "/home/pi/moonraker-telegram-bot-env/lib/python3.9/site-packages/telegram/bot.py", line 133, in decorator
    result = func(*args, **kwargs)
  File "/home/pi/moonraker-telegram-bot-env/lib/python3.9/site-packages/telegram/bot.py", line 5643, in set_my_commands
    result = self._post('setMyCommands', data, timeout=timeout, api_kwargs=api_kwargs)
  File "/home/pi/moonraker-telegram-bot-env/lib/python3.9/site-packages/telegram/bot.py", line 298, in _post
    return self.request.post(
  File "/home/pi/moonraker-telegram-bot-env/lib/python3.9/site-packages/telegram/utils/request.py", line 361, in post
    result = self._request_wrapper(
  File "/home/pi/moonraker-telegram-bot-env/lib/python3.9/site-packages/telegram/utils/request.py", line 279, in _request_wrapper
    raise BadRequest(message)
telegram.error.BadRequest: Bot_command_invalid
Stack (most recent call last):
  File "/home/pi/moonraker-telegram-bot/bot/main.py", line 64, in handle_exception
    logger.error(

### What branch does this occur on?

master

### Reports only on latest master or development branches!

- [X] I have updated to latest development/master version before submitting the bug

aka13-404, 2022-06-15:
Would you kindly attach the whole log as a file?

tekgamerpt, 2022-06-15:
> Would you kindly attach the whole log as a file?

The telegram.log file? Is the same i posted. Always the same errors for each sesson

aka13-404, 2022-06-15:
> > Would you kindly attach the whole log as a file?
> 
> The telegram.log file? Is the same i posted. Always the same errors for each session

I appreciate your insight, but I do need the full log to understand what exactly is happening, not only the place with the error.

tekgamerpt, 2022-06-15:
[telegram.log](https://github.com/nlef/moonraker-telegram-bot/files/8912656/telegram.log)

Here it goes, i just deleted the token and chat id

aka13-404, 2022-06-17:
This must be some 1/0 magic. I don't understand, what is happening and why. I am sorry, I have to meaningful answer for you. Perhaps try a reinstall, maybe it's a library issue?

tekgamerpt, 2022-06-17:
already done that... Sadly didn't worked.
Well... if i find something, ill post it here

aka13-404, 2022-06-28:
@tekgamerpt Can you switch to the dev branch? Nlef has implemented a couple of checks, which might solve or at least show your issue.

tekgamerpt, 2022-06-28:
Hello, sure i'll give it a try

capnkrunchy, 2022-07-01:
I am also having the same issue. I switched to the development branch as recommended and re-installed and did not have any success. The /status command works but regardless of my progress_notification settings (time: 60, also tried time: 5) it will never send an update on its own. I've attached my telegram.log file.

[telegram.log](https://github.com/nlef/moonraker-telegram-bot/files/9024510/telegram.log)

Sungray, 2022-07-25:
Hi, same stuff here, I'm not getting any updates but the bot seems to work otherwise.

tekgamerpt, 2022-07-28:
Hi again, sorry for delay...
After some updates using kiauh, problem remains but in a different way. Sometimes works on first print and after no notifications again.
I think that might have something with telegram bot itself. Maybe it need /bot_restart command to make it work again.
I haven't been printing much these days but if someone can test this i think it would be usefull. Use the bot_restart command before each print and see if anything changes.

Cheers

sapista, 2022-08-02:
Same here, since last update, I'm not getting progress notifications.

aka13-404, 2022-08-03:
We are trying to find what exactly is causing this, but we simply can't reproduce. 
We had a couple of reports that a reinstall helped. Not sure what is happening.

sapista, 2022-08-12:
Not sure how exactly proceed to reinstall it. I just deleted the two corresponding folders for the bot and started over again. The problem is still there: no notifications on telegram.
Can you provide more details on how to reinstall? Thanks

leofauree, 2022-08-13:
I am also suffering from the same issue... Somehow have errors in the journalctl but nothing after that. 
If I change the telegram.conf, nothing changes on the status. I get 0 "auto" notifications but if I send a /status, it comes up... 

My journalctl goes as is: 
[image]

tekgamerpt, 2022-08-28:
Hello, after configurating moonraker.conf to export info to a mqtt server i noticed that the only info that moonraker sends to mqtt is the printer online and offline status, the same thing that telegram bot notify.
So, it seems that is a moonraker problem and not a telegram bot problem has i was thinking. Dunno if it helps

aka13-404, 2022-08-28:
> Hello, after configurating moonraker.conf to export info to a mqtt server i noticed that the only info that moonraker sends to mqtt is the printer online and offline status, the same thing that [...]

Do you have m73 output to your gcode?

tekgamerpt, 2022-08-28:
yes i have

closed 2022-09-17 by aka13-404

spider7611, 2023-12-14:
Have the same problem. No progress notification. Everything else working like /status. What can I do?