nlef/moonraker-telegram-bot #180: [Bug]: Не снимает таймлапсы + не обновляет статус печати

kpp39, 2022-06-24:
### What happened and in what context?

Загрузил клиппер, запустил печать. Статус обновляется только вручную, хотя в конфиге стоит и по таймеру, и по высоте, и по процентам. Также не происходит запись таймлапса. Лог прилагю

### If needed, paste the relevant bot log contents here.

Current Monraker telegram bot config
[bot]
server = localhost
bot_token = 
chat_id = 

[camera]
host = http://localhost:8080/?action=stream
fps = 25

[progress_notification]
percent = 10
height = 10
time = 600

[timelapse]
cleanup = true
height = 0.2
time = 5
target_fps = 15

2022-06-24 08:06:07,887 - klippy - WARNING - Empty thumbnail_path
2022-06-24 08:06:19,395 - __main__ - ERROR - 'dict' object has no attribute 'json'
2022-06-24 09:46:52,643 - camera - WARNING - Reading video frames queue exception 
2022-06-24 09:46:52,679 - camera - WARNING - Reading video frames queue exception 
2022-06-24 09:46:52,715 - camera - WARNING - Reading video frames queue exception 
2022-06-24 09:46:52,754 - camera - WARNING - Reading video frames queue exception 
2022-06-24 09:46:52,794 - camera - WARNING - Reading video frames queue exception 
2022-06-24 09:46:52,838 - camera - WARNING - Reading video frames queue exception 
2022-06-24 09:46:52,880 - camera - WARNING - Reading video frames queue exception 
2022-06-24 09:46:52,916 - camera - WARNING - Reading video frames queue exception 
2022-06-24 09:46:52,956 - camera - WARNING - Reading video frames queue exception 
2022-06-24 09:46:53,004 - camera - WARNING - Reading video frames queue exception 
2022-06-24 09:46:53,045 - camera - WARNING - Reading video frames queue exception 
2022-06-24 09:46:53,090 - camera - WARNING - Reading video frames queue exception 
2022-06-24 09:46:53,129 - camera - WARNING - Reading video frames queue exception 
2022-06-24 09:46:53,168 - camera - WARNING - Reading video frames queue exception 
2022-06-24 09:46:53,208 - camera - WARNING - Reading video frames queue exception 
2022-06-24 09:46:53,248 - camera - WARNING - Reading video frames queue exception 
2022-06-24 09:46:53,287 - camera - WARNING - Reading video frames queue exception 
2022-06-24 09:46:53,330 - camera - WARNING - Reading video frames queue exception 
2022-06-24 09:46:53,370 - camera - WARNING - Reading video frames queue exception 
2022-06-24 09:46:53,409 - camera - WARNING - Reading video frames queue exception 
2022-06-24 09:46:53,450 - camera - WARNING - Reading video frames queue exception 
2022-06-24 09:46:53,490 - camera - WARNING - Reading video frames queue exception 
2022-06-24 09:46:53,529 - camera - WARNING - Reading video frames queue exception 
2022-06-24 09:46:53,568 - camera - WARNING - Reading video frames queue exception 
2022-06-24 09:46:53,605 - camera - WARNING - Reading video frames queue exception 
2022-06-24 09:46:53,688 - camera - WARNING - Reading video frames queue exception 
2022-06-24 09:46:53,727 - camera - WARNING - Reading video frames queue exception 
2022-06-24 09:46:53,765 - camera - WARNING - Reading video frames queue exception 
2022-06-24 09:46:53,809 - camera - WARNING - Reading video frames queue exception 
2022-06-24 09:46:53,849 - camera - WARNING - Reading video frames queue exception 
2022-06-24 09:46:53,895 - camera - WARNING - Reading video frames queue exception 
2022-06-24 09:46:53,957 - camera - WARNING - Reading video frames queue exception 
2022-06-24 09:46:54,008 - camera - WARNING - Reading video frames queue exception 
2022-06-24 09:46:54,055 - camera - WARNING - Reading video frames queue exception 
2022-06-24 09:46:54,141 - camera - WARNING - Reading video frames queue exception 
2022-06-24 09:46:54,204 - camera - WARNING - Reading video frames queue exception 
2022-06-24 09:46:54,274 - camera - WARNING - Reading video frames queue exception 
2022-06-24 09:46:56,913 - camera - WARNING - Writing video frames queue exception <built-in method with_traceback of Full object at 0xa0f9a4a8>
2022-06-24 09:46:57,024 - camera - WARNING - Writing video frames queue exception <built-in method with_traceback of Full object at 0xa0f9a598>
2022-06-24 11:12:01,071 - __main__ - ERROR - Job 6340617d9d4542918902d924c5619fb1 raised[Errno 2] No such file or directory: '/tmp/timelapse/foot_V2_45mm_2h18m_0,20mm_225C_PETG_ENDER3PRO.gcode_2022-06-24_08-06/lapse.lock'
Traceback (most recent call last):
[... 155 lines cut ...]
  File "/home/klipper/moonraker-telegram-bot-env/lib/python3.9/site-packages/telegram/ext/extbot.py", line 226, in get_updates
    updates = super().get_updates(
  File "/home/klipper/moonraker-telegram-bot-env/lib/python3.9/site-packages/telegram/bot.py", line 133, in decorator
    result = func(*args, **kwargs)
  File "/home/klipper/moonraker-telegram-bot-env/lib/python3.9/site-packages/telegram/bot.py", line 3057, in get_updates
    self._post(
  File "/home/klipper/moonraker-telegram-bot-env/lib/python3.9/site-packages/telegram/bot.py", line 298, in _post
    return self.request.post(
  File "/home/klipper/moonraker-telegram-bot-env/lib/python3.9/site-packages/telegram/utils/request.py", line 361, in post
    result = self._request_wrapper(
  File "/home/klipper/moonraker-telegram-bot-env/lib/python3.9/site-packages/telegram/utils/request.py", line 265, in _request_wrapper
    raise NetworkError(f'urllib3 HTTPError {error}') from error
telegram.error.NetworkError: urllib3 HTTPError HTTPSConnectionPool(host='api.telegram.org', port=443): Max retries exceeded with url: /bot5316443677:AAELne9Ur181Uv8iv1mwwxFwu6IiumO4rgU/getUpdates (Caused by ConnectTimeoutError(<telegram.vendor.ptb_urllib3.urllib3.connectionpool.HTTPSConnectionPool object at 0xa31a8220>, 'Connect timed out. (connect timeout=5.0)'))
2022-06-24 20:12:36,507 - __main__ - ERROR - 'dict' object has no attribute 'json'
2022-06-24 20:22:30,484 - __main__ - ERROR - 'gcode_position'
2022-06-24 20:22:31,488 - __main__ - ERROR - 'gcode_position'
2022-06-24 20:36:36,617 - __main__ - ERROR - Unknown state: cancelled

### What branch does this occur on?

master

### Reports only on latest master or development branches!

- [X] I have updated to latest development/master version before submitting the bug

aka13-404, 2022-06-24:
Дядь, ты помнишь что я просил? Приложи весь лог файлом

kpp39, 2022-06-24:
> Дядь, ты помнишь что я просил? Приложи весь лог файлом
вот файл. я его весь и приложил, только не файлом)
[telegram.log](https://github.com/nlef/moonraker-telegram-bot/files/8980418/telegram.log)

fcps3, 2023-04-13:
Someone solved this problem? Same here...