nlef/moonraker-telegram-bot #186: Timelapse snapshots on layer change

SharkWipf, 2022-07-20:
### What would you like to see added?

Currently, timelapse snapshots are time-based, and the result, depending on how your camera is set up, can be all over the place and messy.
Layer changes are often around the same locations every layer, relative to each other (possibly depending on your seam settings).
I don't know if it's possible to "hook into" gcode being run (I don't know how the live gcode viewer does it), but if it were possible to only run snapshots on layer changes, that should result in much cleaner timelapses I'd think.

aka13-404, 2022-07-20:
I would appreciate it if you at least pretended to have read the docs before you open a ticket.
Please refer to the [sample config](https://github.com/nlef/moonraker-telegram-bot/wiki/Sample-config) to learn what different timelapse method exist out of the box, and the [corresponding section](https://github.com/nlef/moonraker-telegram-bot/wiki/Interacting-with-klipper#controlling-timelapse-parameters-via-gcode) in g-code commands on how to manually control the lapsing for all the needs you may have.

closed 2022-07-20 by aka13-404

SharkWipf, 2022-07-20:
Believe it or not, I've been through the documentation several times, especially the sample config page, I even looked at the manual mode timelapse config option, but couldn't tell from that that such a thing was possible.  
I did not expect to find it back in the "Interacting with Klipper" section, a page I've also read more than once, but not specifically looking for this setting, so I read over it.

The fact that I read over it in the documentation is my bad, and I apologize for that, but there's no need to get rude about it and assume I didn't read the documentation. It's not like you're getting dozens of issues per day that you have to sift through.