nlef/moonraker-telegram-bot #188: multiple chat_id support - for multiple users

pejotigrek, 2022-07-21:
### What would you like to see added?

I already successfuly connected my bot with moonraker installation, but was wondering if it is possible to send notifications to multiple users?

we've got few people that use the printer, it would be great to be able to send info to multiple chat_ids so everybody can react if something is wrong with the print.

aka13-404, 2022-07-27:
Multiple chats won't be possible for now. What you can do, is use groups. Have you seen the group parameter?

pejotigrek, 2022-07-27:
> Multiple chats won't be possible for now. What you can do, is use groups. Have you seen the group parameter?

yes, I saw the group options, but didn't wanted to go that way - long story short: didn't wanted to multiply chats for others and for me myself [too many "important" ones ;(]

by "for now" you mean "maybe in the [not so] distant future" or "probably never"? ;) asking just to know if I need to think about workaround ;)

aka13-404, 2022-08-03:
This is more propable a "never" or a "very distant future" thing, since what this would imply is user rights, multiple users entering commands simultaniously etc. It's a very ugly can of worms we are not yet willing to open :(.

pejotigrek, 2022-08-03:
> This is more propable a "never" or a "very distant future" thing, since what this would imply is user rights, multiple users entering commands simultaniously etc. It's a very ugly can of worms we are not yet willing to open :(.

understood. I'll keep my fingers crossed, but for now will try to think about the workaround.

closed 2022-08-03 by pejotigrek