nlef/moonraker-telegram-bot #189: Not installed

kobzzon, 2022-07-30:
dist.run_commands()
    File "/usr/lib/python3/dist-packages/setuptools/_distutils/dist.py", line 967, in run_commands
      self.run_command(cmd)
    File "/usr/lib/python3/dist-packages/setuptools/_distutils/dist.py", line 986, in run_command
      cmd_obj.run()
    File "/usr/lib/python3/dist-packages/wheel/bdist_wheel.py", line 259, in run
      self.run_command('install')
    File "/usr/lib/python3/dist-packages/setuptools/_distutils/cmd.py", line 313, in run_command
      self.distribution.run_command(command)
    File "/usr/lib/python3/dist-packages/setuptools/_distutils/dist.py", line 986, in run_command
      cmd_obj.run()
    File "/usr/lib/python3/dist-packages/setuptools/command/install.py", line 61, in run
      return orig.install.run(self)
    File "/usr/lib/python3/dist-packages/setuptools/_distutils/command/install.py", line 578, in run
      self.run_command(cmd_name)
    File "/usr/lib/python3/dist-packages/setuptools/_distutils/cmd.py", line 313, in run_command
      self.distribution.run_command(command)
    File "/usr/lib/python3/dist-packages/setuptools/_distutils/dist.py", line 985, in run_command
      cmd_obj.ensure_finalized()
    File "/usr/lib/python3/dist-packages/setuptools/_distutils/cmd.py", line 107, in ensure_finalized
      self.finalize_options()
    File "/usr/lib/python3/dist-packages/setuptools/command/install_lib.py", line 17, in finalize_options
      self.set_undefined_options('install',('install_layout','install_layout'))
    File "/usr/lib/python3/dist-packages/setuptools/_distutils/cmd.py", line 290, in set_undefined_options
      setattr(self, dst_option, getattr(src_cmd_obj, src_option))
    File "/usr/lib/python3/dist-packages/setuptools/_distutils/cmd.py", line 103, in getattr
      raise AttributeError(attr)
  AttributeError: install_layout
  ----------------------------------------
  ERROR: Failed building wheel for tornado
Failed to build ujson tornado
ERROR: Could not build wheels for ujson, tornado which use PEP 517 and cannot be installed directly

aka13-404, 2022-08-03:
https://github.com/nlef/moonraker-telegram-bot/issues/183

closed 2022-10-30 by nlef

KLXYinc, 2024-01-18:
i got the same problem with the newest version