nlef/moonraker-telegram-bot #191: Interactive buttons for MSG_RESPOND= type of replies

aka13-404, 2022-08-24:
### What would you like to see added?

It would be cool not only to provide "highlighted" macros as  in https://github.com/nlef/moonraker-telegram-bot/wiki/Ideas-for-macros , but also to generate a keyboard with labels based on some specific prefix. 

Might be an interesting addition for gui-like experience with SCREW_TILT_ADJUST for example, so that you get a message with "ACCEPT, ADJUST, ..." which can be clicked and would then send a gcode to klipper.

nlef, 2023-01-02:
New prefix `tgcustom_keyboard` added. The full foramt of the command is shown below.
`
RESPOND PREFIX=tgcustom_keyboard MSG="message='Some Message', buttons=[{name='z-.01', command='TESTZ Z=-.1'},{name='z-.05', command='TESTZ Z=-.5'}],[{name='z+.01', command='TESTZ Z=.1'},{name='z+.05', command='TESTZ Z=.5'}],[{name='CleanUp', command='delete'}]"
`

aka13-404, 2023-01-15:
1. Autodeletion flag for selecting any button in a message
2. Syntaxis propably needs to be fixed/changed.
Currently working is the following scenario,due to how jinja parses stuff:
```
{% set bot_msg =  ""message='Some Message', buttons=[{name='z-.01', command='TESTZ Z=-.1'},{name='z-.05', command='TESTZ Z=-.5'}],[{name='z+.01', command='TESTZ Z=.1'},{name='z+.05', command='TESTZ Z=.5'}],[{name='CleanUp', command='delete'}]" %}
RESPOND PREFIX=tgcustom_keyboard MSG="{bot_msg}"
```

3. Adding spaces to "message = 'some_message'

closed 2023-02-05 by aka13-404