nlef/moonraker-telegram-bot #193: RESPOND PREFIX=tgnotify MSG not working

pertoft, 2022-08-26:
Hi,

First of all, amazing work you are doring :-)

Next, i have been using the bot for a long time, but at some point the RESPOND command broke and the bot stopped presenting the messages on telegram from Klipper respond commands
Example from klipper:
`16:30:27 
$ RESPOND PREFIX=tgnotify MSG=my_message
16:30:27 
tgnotify my_message`

Shoud produce a telegram message, but nothing happens.

Everything else is working e.g. /status, /video, printer online etc. 

`pi@raspberrypi:~/klipper_config $ sudo journalctl -f -u moonraker-telegram-bot
-- Logs begin at Fri 2022-08-26 14:48:02 BST. --
Aug 26 15:29:25 raspberrypi python[6702]: 2022-08-26 15:29:25,804 - camera - WARNING - Reading video frames queue exception
Aug 26 15:29:25 raspberrypi python[6702]: 2022-08-26 15:29:25,916 - camera - WARNING - Reading video frames queue exception
Aug 26 15:29:26 raspberrypi python[6702]: 2022-08-26 15:29:26,033 - camera - WARNING - Reading video frames queue exception
Aug 26 15:29:26 raspberrypi python[6702]: 2022-08-26 15:29:26,126 - camera - WARNING - Reading video frames queue exception
Aug 26 15:29:26 raspberrypi python[6702]: 2022-08-26 15:29:26,237 - camera - WARNING - Reading video frames queue exception
Aug 26 15:30:07 raspberrypi systemd[1]: Stopping Starts Moonraker Telegram Bot on startup...
Aug 26 15:30:07 raspberrypi systemd[1]: moonraker-telegram-bot.service: Main process exited, code=killed, status=15/TERM
Aug 26 15:30:07 raspberrypi systemd[1]: moonraker-telegram-bot.service: Succeeded.
Aug 26 15:30:07 raspberrypi systemd[1]: Stopped Starts Moonraker Telegram Bot on startup.
Aug 26 15:30:07 raspberrypi systemd[1]: Started Starts Moonraker Telegram Bot on startup.
Aug 26 15:30:11 raspberrypi python[7408]: 2022-08-26 15:30:11,072 - camera - DEBUG -
Aug 26 15:30:11 raspberrypi python[7408]: General configuration for OpenCV 3.2.0 =====================================
Aug 26 15:30:11 raspberrypi python[7408]:   Version control:               unknown
Aug 26 15:30:11 raspberrypi python[7408]:   Extra modules:
Aug 26 15:30:11 raspberrypi python[7408]:     Location (extra):            /build/opencv-L65chJ/opencv-3.2.0+dfsg/contrib/modules
Aug 26 15:30:11 raspberrypi python[7408]:     Version control (extra):     unknown
Aug 26 15:30:11 raspberrypi python[7408]:   Platform:
Aug 26 15:30:11 raspberrypi python[7408]:     Timestamp:                   2019-02-12T11:26:31Z
Aug 26 15:30:11 raspberrypi python[7408]:     Host:                        Linux 4.4.0-124-generic armv8l
Aug 26 15:30:11 raspberrypi python[7408]:     CMake:                       3.13.4
Aug 26 15:30:11 raspberrypi python[7408]:     CMake generator:             Unix Makefiles
Aug 26 15:30:11 raspberrypi python[7408]:     CMake build tool:            /usr/bin/make
Aug 26 15:30:11 raspberrypi python[7408]:     Configuration:               Release
Aug 26 15:30:11 raspberrypi python[7408]:   C/C++:
Aug 26 15:30:11 raspberrypi python[7408]:     Built as dynamic libs?:      YES
Aug 26 15:30:11 raspberrypi python[7408]:     C++ Compiler:                /usr/bin/c++  (ver 8.2.0)
Aug 26 15:30:11 raspberrypi python[7408]:     C++ flags (Release):         -g -O2 -fdebug-prefix-map=/build/opencv-L65chJ/opencv-3.2.0+dfsg=. -fstack-protector-strong -Wformat -Werror=format-security -Wdate-time -D_FORTIFY_SOURCE=2   -fsigned-char -W -Wall -Werror=return-type -Werror=non-virtual-dtor -Werror=address -Werror=sequence-point -Wformat -Werror=format-security -Wmissing-declarations -Wundef -Winit-self -Wpointer-arith -Wshadow -Wsign-promo -Wno-narrowing -Wno-delete-non-virtual-dtor -Wno-comment -fdiagnostics-show-option -pthread -fomit-frame-pointer -mfp16-format=ieee -ffunction-sections -fvisibility=hidden -fvisibility-inlines-hidden -g -O2 -fdebug-prefix-map=/build/opencv-L65chJ/opencv-3.2.0+dfsg=. -fstack-protector-strong -Wformat -Werror=format-security  -DNDEBUG
Aug 26 15:30:11 raspberrypi python[7408]:     C++ flags (Debug):           -g -O2 -fdebug-prefix-map=/build/opencv-L65chJ/opencv-3.2.0+dfsg=. -fstack-protector-strong -Wformat -Werror=format-security -Wdate-time -D_FORTIFY_SOURCE=2   -fsigned-char -W -Wall -Werror=return-type -Werror=non-virtual-dtor -Werror=address -Werror=sequence-point -Wformat -Werror=format-security -Wmissing-declarations -Wundef -Winit-self -Wpointer-arith -Wshadow -Wsign-promo -Wno-narrowing -Wno-delete-non-virtual-dtor -Wno-comment -fdiagnostics-show-option -pthread -fomit-frame-pointer -mfp16-format=ieee -ffunction-sections -fvisibility=hidden -fvisibility-inlines-hidden -g  -O0 -DDEBUG -D_DEBUG
Aug 26 15:30:11 raspberrypi python[7408]:     C Compiler:                  /usr/bin/cc
Aug 26 15:30:11 raspberrypi python[7408]:     C flags (Release):           -g -O2 -fdebug-prefix-map=/build/opencv-L65chJ/opencv-3.2.0+dfsg=. -fstack-protector-strong -Wformat -Werror=format-security -Wdate-time -D_FORTIFY_SOURCE=2   -fsigned-char -W -Wall -Werror=return-type -Werror=non-virtual-dtor -Werror=address -Werror=sequence-point -Wformat -Werror=format-security -Wmissing-declarations -Wmissing-prototypes -Wstrict-prototypes -Wundef -Winit-self -Wpointer-arith -Wshadow -Wno-narrowing -Wno-comment -fdiagnostics-show-option -pthread -fomit-frame-pointer -mfp16-format=ieee -ffunction-sections -fvisibility=hidden -g -O2 -fdebug-prefix-map=/build/opencv-L65chJ/opencv-3.2.0+dfsg=. -fstack-protector-strong -Wformat -Werror=format-security  -DNDEBUG
Aug 26 15:30:11 raspberrypi python[7408]:     C flags (Debug):             -g -O2 -fdebug-prefix-map=/build/opencv-L65chJ/opencv-3.2.0+dfsg=. -fstack-protector-strong -Wformat -Werror=format-security -Wdate-time -D_FORTIFY_SOURCE=2   -fsigned-char -W -Wall -Werror=return-type -Werror=non-virtual-dtor -Werror=address -Werror=sequence-point -Wformat -Werror=format-security -Wmissing-declarations -Wmissing-prototypes -Wstrict-prototypes -Wundef -Winit-self -Wpointer-arith -Wshadow -Wno-narrowing -Wno-comment -fdiagnostics-show-option -pthread -fomit-frame-pointer -mfp16-format=ieee -ffunction-sections -fvisibility=hidden -g  -O0 -DDEBUG -D_DEBUG
Aug 26 15:30:11 raspberrypi python[7408]:     Linker flags (Release):      -Wl,-z,relro -Wl,-z,now -Wl,-z,relro -Wl,-z,now
Aug 26 15:30:11 raspberrypi python[7408]:     Linker flags (Debug):        -Wl,-z,relro -Wl,-z,now
Aug 26 15:30:11 raspberrypi python[7408]:     ccache:                      NO
Aug 26 15:30:11 raspberrypi python[7408]:     Precompiled headers:         NO
Aug 26 15:30:11 raspberrypi python[7408]:     Extra dependencies:          /usr/lib/arm-linux-gnueabihf/libwebp.so gdcmMSFF /usr/lib/arm-linux-gnueabihf/libImath.so /usr/lib/arm-linux-gnueabihf/libIlmImf.so /usr/lib/arm-linux-gnueabihf/libIex.so /usr/lib/arm-linux-gnueabihf/libHalf.so /usr/lib/arm-linux-gnueabihf/libIlmThread.so /usr/lib/libgdal.so gtk-3 gdk-3 pangocairo-1.0 pango-1.0 atk-1.0 cairo-gobject cairo gdk_pixbuf-2.0 gio-2.0 gobject-2.0 glib-2.0 gthread-2.0 dc1394 avcodec avformat avutil swscale avresample gphoto2 gphoto2_port /usr/lib/arm-linux-gnueabihf/hdf5/serial/libhdf5.so /usr/lib/arm-linux-gnueabihf/libpthread.so /usr/lib/arm-linux-gnueabihf/libsz.so /usr/lib/arm-linux-gnueabihf/libdl.so /usr/lib/arm-linux-gnueabihf/libm.so vtkRenderingOpenGL vtkImagingHybrid vtkIOImage vtkCommonDataModel vtkCommonMath vtkCommonCore vtksys vtkCommonMisc vtkCommonSystem vtkCommonTransforms vtkCommonExecutionModel vtkDICOMParser vtkIOCore /usr/lib/arm-linux-gnueabihf/libz.so vtkmetaio /usr/lib/arm-linux-gnueabihf/libjpeg.so /usr/lib/arm-linux-gnueabihf/libpng.so /usr/lib/arm-linux-gnueabihf/libtiff.so vtkImagingCore vtkRenderingCore vtkCommonColor vtkFiltersExtraction vtkFiltersCore vtkFiltersGeneral vtkCommonComputationalGeometry vtkFiltersStatistics vtkImagingFourier vtkalglib vtkFiltersGeometry vtkFiltersSources vtkInteractionStyle vtkRenderingLOD vtkFiltersModeling vtkIOPLY vtkIOGeometry vtkFiltersTexture vtkRenderingFreeType /usr/lib/arm-linux-gnueabihf/libfreetype.so vtkftgl vtkIOExport vtkRenderingAnnotation vtkImagingColor vtkRenderingContext2D vtkRenderingGL2PS vtkRenderingContextOpenGL /usr/lib/arm-linux-gnueabihf/libgl2ps.so vtkRenderingLabel tesseract lept dl m pthread rt tbb
Aug 26 15:30:11 raspberrypi python[7408]:     3rdparty dependencies:
Aug 26 15:30:11 raspberrypi python[7408]:   OpenCV modules:
Aug 26 15:30:11 raspberrypi python[7408]:     To be built:                 core flann hdf imgproc ml photo reg surface_matching video viz freetype fuzzy imgcodecs shape videoio highgui objdetect plot superres ts xobjdetect xphoto [... cut ...]
Aug 26 15:30:11 raspberrypi python[7408]:   File "/home/pi/moonraker-telegram-bot-env/lib/python3.7/site-packages/telegram/utils/request.py", line 279, in _request_wrapper
Aug 26 15:30:11 raspberrypi python[7408]:     raise BadRequest(message)
Aug 26 15:30:11 raspberrypi python[7408]: telegram.error.BadRequest: Bot_command_invalid
Aug 26 15:30:11 raspberrypi python[7408]: Stack (most recent call last):
Aug 26 15:30:11 raspberrypi python[7408]:   File "/home/pi/moonraker-telegram-bot/bot/main.py", line 67, in handle_exception
Aug 26 15:30:11 raspberrypi python[7408]:     stack_info=True,
`

aka13-404, 2022-09-17:
Can you check with latest release please?

blonestar, 2022-09-21:
Then it is an issue, the same one, I guess...
https://github.com/nlef/moonraker-telegram-bot/discussions/201

closed 2023-02-01 by aka13-404