nlef/moonraker-telegram-bot #196: Status message rework, Secrets, QoL fixes

aka13-404, 2022-09-10:
This update breaks the config (sorry). Please check out the release notes if you plan updating.
https://github.com/nlef/moonraker-telegram-bot/releases/tag/v1.4.5

Grafton1046, 2022-09-13:
attempted to setup a secrets file, do the bot_token: and chat_id: fields sill need to exist in the telegram.conf file?

aka13-404, 2022-09-13:
No, of course not, you can skip them in the main telegram.conf

shcad, 2022-09-15:
Please add info to the Wiki that secret file MUST have [secrets] section header included to be parsed properly. Not only main config file.
Currently its not so obvious, and could be misleading (as for me to for a while).

BTW. Thank You for implementing

aka13-404, 2022-09-15:
Oh man, I really did forget to add that, thank you for the catch.
It certainly is not obvious. Appreciate the effort, will fix it ASAP.

closed 2023-02-05 by aka13-404